thp/psmoveapi #150: The exposure settings, they do nothing. PS3Eye tracking only works in low lighting.

cboulay, 2015-02-12:
In OSX, pushing the globe against the iSight camera automatically lowers its exposure, so that works.

When using the PS3Eye in Windows or OSX, tracking only works in low-light conditions. I think many people get this problem. See #141 

in `psmove_tracker_new_with_camera(int camera)`, there is a call to `psmove_tracker_set_exposure(tracker, Exposure_LOW);`

In `psmove_tracker_set_exposure(PSMoveTracker *tracker, enum PSMoveTracker_Exposure exposure)`, `target_luminance` is set depending on the exposure input, with Exposure_LOW yielding `target_luminance = 0;`

Then it calls 
`tracker->exposure = psmove_tracker_adapt_to_light(tracker, target_luminance);`

That function has an algorithm to adapt the exposure setting to yield the target_luminance. When target_luminance==0, it immediately returns with minimum_exposure: 2051;

Then 
`camera_control_set_parameters(tracker->cc, 0, 0, 0, tracker->exposure,
            0, 0xffff, 0xffff, 0xffff, -1, -1);`

For reference, here's the function declaration.
`camera_control_set_parameters(CameraControl* cc, int autoE, int autoG, int autoWB, int exposure, int gain, int wbRed, int wbGreen, int wbBlue, int contrast, int brightness)`

That function is platform specific. In Windows, using the PSEYE, it sets a registry key with these values, then restarts camera capture with openCV. This calls `cvCreateCameraCapture( cameraID );`

As far as I can tell, [this function](https://github.com/Itseez/opencv/blob/61886a61de82ef5e837b1f3d163f4666ea16a2da/modules/videoio/src/cap.cpp#L120) does not care what's in the registry.

In OSX, we just get a warning: `psmove_WARNING("Unimplemented: Setting of PS3EYEDriver parameters\n");`

cboulay, 2015-02-12:
I fixed this problem for Windows in [my own fork](https://github.com/cboulay/psmoveapi/commit/ca8717695773b241823a5d3ba1edff545b6e7744). I made quite a few changes in my own fork that I'm not sure you want to merge upstream. For example, I removed PS3EYEDriver as a submodule.

thp, 2015-05-25:
If you have a fix, I'd still be interested in getting it merged upstream. Any chance we could do that? ;)

peoronoob, 2015-10-27:
I'm having a related issue on Linux. Apparently psmoveapi sets auto exposure to 0 (`v4l2_set_control(fd, V4L2_CID_EXPOSURE_AUTO, autoE);` in `camera_control_set_parameters`, called by `psmove_tracker_set_exposure`, `psmove_tracker_new_with_camera`, `psmove_tracker_new`).

With my camera (PS3Eye, using GSPCA/OV534 USB Camera Driver), having auto exposure set to 0 means enabling the auto exposure. I found out that `V4L2_CID_EXPOSURE_AUTO` needs to be set to 65535 (which is different from `V4L2_EXPOSURE_MANUAL`, WTF?) in order to disable auto exposure.

I don't understand what's going on, and wouldn't know how to fix this issue in a clean way.
I decided to have `camera_control_set_parameters` returning immediately without doing anything, and to manually control the camera settings using `v4l2-ctl` (`v4l2-ctl -d /dev/video0 -c auto_exposure=1` to set manual exposure, `v4l2-ctl -d /dev/video0 -c exposure=10` to control the exposure level etc).

Hope this can help!

procedural, 2017-02-22:
@peoronoob thanks, it did help in my case too!

alessiodeltrecco, 2019-12-21:
Hi, i think i've found a solution, at least for Linux it's working...
After some researches and tests, based on the infos contained in @peoro's reply, i've modified `psmove_tracker.c` changing all the calls to `camera_control_set_parameters` from:

```
camera_control_set_parameters(tracker->cc, 0, 0, 0, tracker->settings.camera_exposure,
		0, 0xffff, 0xffff, 0xffff, -1, -1, tracker->settings.camera_mirror);
```
to: 
```
camera_control_set_parameters
[psmove_tracker.zip](https://github.com/thp/psmoveapi/files/3991939/psmove_tracker.zip)

(tracker->cc, 65535, 0, 0, tracker->settings.camera_exposure,
		0, 0xffff, 0xffff, 0xffff, -1, -1, tracker->settings.camera_mirror);

```
and it worked :heart_eyes: 

Hope this helps and the fix 

could be merged upstream

[psmove_tracker.zip](https://github.com/thp/psmoveapi/files/3991940/psmove_tracker.zip)

commit 40d8c06e (2019-12-30): PSEye on Linux: Fix auto-exposure setting (Fixes #150)
commit 61e6427f (2020-01-01): Merge pull request #399 from thp/psmove-linux-autoexposure-fix

PSEye on Linux: Fix auto-exposure setting (Fixes #150)
closed 2020-01-01 by thp