thp/psmoveapi #155: Run-time detection of camera type.

cboulay, 2015-02-16:
It should be possible to alter the code so the camera type is detected at runtime. This should enable switching camera types between application runs, and it should reduce the number of CMake parameters and preprocessor directives (I hope I'm using that word correctly).
See this [thread](https://groups.google.com/forum/#!topic/psmove/azU9YYGkDmk) for a discussion.

cboulay, 2015-07-02:
I'm not working on this at the moment, but just came across this so I thought I'd write down some thoughts and ask a few questions.

@ronin18 , @Lirrec, I have a couple CL Eye specific questions. As I don't have the driver, I thought maybe you could provide some insight.

I'm thinking of order of preference as follows (may also have a similar ENUM):

* PSEYE_OV534
    * Always available in Linux
* PSEYE_CLREG
    * Always available in Window
    * Uses OpenCV, so how can we test for this vs a regular webcam?
    * Camera is configured via registry
* PSEYE_CLSDK
    * Only available in Windows if CL SDK is configured at compile-time.
    * Is it sufficient to test for this with `CLEyeGetCameraCount > 0`?
    * Camera is configured via CL SDK
* PSEYE_LIBUSB
    * Only available in Win or Mac if libusb-1.0 & PS3EYEDriver are configured at compile-time.
    * Can be tested by examining enumerated libusb devices
    * Camera is configured via PS3EYEDriver
* ISIGHT
    * Always available in Mac
    * Special initialization sequence because of auto-luminance trickery.
* WEBCAM
    * Always available.
    * Uses OpenCV but has some platform-specific quirks.

For those of us that make binaries available, we should always compile with as many options as possible. Then, during run-time tracker initialization, some checks will be done against each of these options and the first match will be picked and camera type will be stored. From then on, all setup, capture, uninitialize functions will use if/else or switches based on the camera type to choose the correct code path. In the extreme case, in Windows there will be PSEYE_CLREG, PSEYE_CLSDK, PSEYE_LIBUSB and WEBCAM options available at runtime. For Mac there will be PSEYE_LIBUSB, ISIGHT, and WEBCAM. For Linux there is PSEYE_OV534 (maybe not necessary) and WEBCAM; they both use the same interface.

Obviously we can't make the CL Eye SDK a hard dependency (because it's non-free). Do we still want to support it? Does it have any features that the other CL Eye driver (non-SDK) or the PS3EYEDriver does not?

ronin18, 2015-07-03:
The CL Driver is also not free. Only cheaper.
I'm not aware of any differences __for the user__ besides the driver can only use one PSeye per computer.

In this discussion we really have to distinguish between developers and users.
I'm wondering if binaries, build with the CL SDK, can be used by users with only the CL Driver available.
Usually (compared to other SDKs I know) this would be the case.
If so, we (the developers) could just use only the CL SDK (with the other available free options) and tell users to buy a copy of the CL Drivers.
Though this would force developers to buy the horrendously expensive 4.99$ SDK instead of the 2.99$ Driver.
If this is not the case I agree with cboulay to distinguish between them, since there are few differences for the project at the moment.

In the matter of runtime detection I have no opinion, since I'm only building for a specific application.

thp, 2016-12-28:
Now that PS3EYEDriver is available, do we really need CL SDK and stuff? In fact, CLEye SDK support has been removed, we only support the CL Driver via the registry hacks (and maybe that can go as well?).

cboulay, 2016-12-28:
It's not really necessary. You'd be leaving out the people that want to use the PS3Eye as a DirectShow device (e.g., for Skype) AND want to use psmoveapi. I imagine this group is quite small.

PS3EYEDriver now has some improvements so its performance is superior to CL Driver (this was not always the case), so it's probably acceptable to demand people use PS3EYEDriver then you can drop support for CL Driver.

thp, 2023-01-03:
This is now "fixed" by cleaning up the camera support. On Windows and macOS, PS3EYEDriver is used. On Linux, there is now "native" support for PS3, PS4 and PS5 cameras via V4L2 since #472, including detecting and configuring the camera type. #475 will add support for querying the camera metadata.

closed 2023-01-03 by thp