thp/psmoveapi #157: What is the purpose of the swapped axes?

ccoenen, 2015-02-22:
In [psmove_orientation, where the AHRS Code is used](https://github.com/thp/psmoveapi/blob/682e035c88db8f7a1837240f7c75510fbd650fc7/src/psmove_orientation.c#L144-156) there's an odd piece of code that i don't understand (highlighted in previous link).

Basically all the axes are swapping Y and Z (or at least that's my interpretation of the 0, 2, 1 order). I believe that this causes Weirdness further along the road, for example in the orientation code of the UniMove plugin.

My understanding (we looked up the spec sheets of the acceelerometer) is, that the controllers Y-Axis is the one along its longest side. The Sphere being +Y and the usb-connector being -Y. The Z-Axis goes through the PS-Button on the front and the barcode-sticker on the back. And X should be the remaining "left-right" axis.

In my personal project, i swapped these indices to be "in order" [0,1,2]... and everything else fell into place.

Would this be a valuable pull request for you, or is there something else to it, that i don't quite understand?

ccoenen, 2015-02-24:
I'm referring to these changes here, that i need to get the axes into unity correctly:
https://github.com/ccoenen/UniMove/commit/ecd10c293926d862600009df7b87f232a76d1fa3

ccoenen, 2015-03-03:
the described changes can be seen here: [psmove_orientation.c](https://github.com/ccoenen/psmoveapi/commit/91bef47b2922de4a5a41a34bd8bed0e43f4d39e1#diff-6c51c8221db66c6c2f0297a3e2c3d59d)

cboulay, 2015-05-18:
I've been looking at this recently and I think I can guess at the answer.

The raw sensor readings have a left-handed coordinate system. +X is from Select to Start, +Y is from USB to sphere, and +Z is from PSMove to trigger. So if we set the controller upright (sphere to the sky) with the PSMove button facing the camera, then the coordinate system is consistent with the positional coordinate system from the camera.

However, Thomas likes to turn on mirroring around the Y-axis (+X is to the right when facing the camera), and he likes the default PSMove orientation to be pointing at the camera. By keeping X as is, setting Y-sensor to be -Z, and setting Z-sensor to be  -Y, then the orientation coordinate system matches the tracker coordinate system when the PSMove is facing the camera (Move button up) and mirroring is set to on.

By turning on camera mirroring, the default LHS becomes a RHS. Unity and UE4 both use LHS. We can turn off mirroring and then we get the LHS back for position, but to undo the changes for rotation is more difficult.

You may have noticed that psmove_enable_orientation calls psmove_reset_orientation, and most of the examples that use orientation hook a button press into psmove_reset_orientation. psmove_reset_orientation is just a wrapper around [psmove_reset_quaternion](https://github.com/thp/psmoveapi/blob/bd3a44666b6928bf3948644db7fac0fa61b32f8b/src/psmove_orientation.c#L228-L246). That function calculates the unit conjugate (i.e., inverse) of the present quaternion, described as the [quaternion when the controller is pointing at the camera](https://github.com/thp/psmoveapi/blob/bd3a44666b6928bf3948644db7fac0fa61b32f8b/src/psmove_orientation.c#L61) then subsequent calls to get the orientation will multiply the new raw quaternion by this unit conjugate.

IMO this is a little too ham-fisted as a default behaviour that can't be undone. I think the only orientation reset that needs to be done is the yaw, because tilt (pitch & roll) are real physical things based on gravity, whereas yaw=0 is rather arbitrary unless you're playing a compass game. This probably doesn't need to be zeroed by default (maybe someone wants to make a compass game!), but a convenience function to handle yaw-zeroing would probably be generally useful, as would positional zeroing.

I'm trying to make these changes to my fork but it's not working exactly as I'd hoped. Do you happen to know if Madgwick's algorithm relies on a certain axis being the gravity axis?

cboulay, 2015-05-18:
I had the Z-direction negated in that last post. I've since updated it.

ccoenen, 2015-05-18:
as far as i can tell, the Madgwick Algorithm works regardless of what's what. I just swapped axes, because the "standing upright" position translated best to the thing we were doing. And also because it seems to be the way that the sensors are mounted internally.

I guess that even my other work in progress (the dead reckoning) would not really prefer any particular ordering of axes or handedness, as long as it's consistent.

thp, 2015-05-25:
AFAIR the axis alignment still has to match up between the accelerometer, magnetometer and gyroscope (at least be in the same relationship to each other as on Madgwick's IMU device). Accelerometer and Magnetometer are easy, because they both measure directional things, whereas the magnetometer measures rotational things.

But the order of the axes shouldn't have any significance (you will just get the quaternion output in a different direction and need to rotate that). The axis alignment in the API has been done so it behaves well for AR use cases (camera pointing towards the user, model superimposed on the mirrored camera image).

To not break existing code, I'm for not changing this as a default, but it might be worthwhile doing it as an additional run-time API call. Note that there's already `psmove_tracker_set_mirror()`, and we do convert the modelview matrix accordingly if it's set: https://github.com/thp/psmoveapi/blob/master/src/tracker/psmove_fusion.cpp#L93

cboulay, 2015-05-25:
But it's still arbitrary that the default controller orientation is with its long axis being horizontal (i.e. sphere pointing at camera). If you had instead decided that the default orientation was with the controller pointing straight up then you wouldn't have to swap the axes.

There's an argument to be made for either case. Starting with the sphere pointing toward the camera gives you the most rotational freedom before the controller starts to occlude the sphere. Starting with the sphere pointing up means that you can ask the user to set the controller on a flat surface, resting on the end with the USB connector, and use that as a zeroed orientation (also a good time to measure sensor noise).

I plan to revise the orientation code at some point because I'm getting pretty bad yaw drift. The [moveframework](https://code.google.com/p/moveframework/) has some interesting solutions and I've read some other useful tips in the [Oculus blog](https://www.oculus.com/blog/category/tech-talk/) (see posts by Steve LaValle, from oldest to newest). I think a necessary step will be to ask users to place the controller in a known orientation while sensor noise and offsets are measured. Because of that, I think it might be better to have the zero-orientation be with the sphere pointing straight up. But I'll spend some time working on this before I make an argument to change the code.

thp, 2015-05-29:
Sounds good, I'd be happy to accept patches that improve the orientation tracking. The reason for having the initial position is pointing to the camera is that in PS3 games it's done like that, and if it's pointing straight up, getting the rotation around the gravity axis (from the magnetometer) is not really possible.

cboulay, 2015-07-04:
This is a bit tangential, but I just came across something similar / relevant that I hope @thp can shed some light on.

Some [code from psmove_fusion](https://github.com/thp/psmoveapi/blob/master/src/tracker/psmove_fusion.cpp#L92-L98):

```
float q0, q1, q2, q3;
psmove_get_orientation(move, &q0, &q1, &q2, &q3);
if (psmove_tracker_get_mirror(fusion->tracker)) {
    /* Need to invert these two axes if mirroring is enabled */
    q3 *= -1.;
    q2 *= -1.;
}
glm::quat quaternion(q3, q2, q1, q0);
```

According to [psmove_get_orientation](https://github.com/thp/psmoveapi/blob/master/src/psmove.c#L1912-L1913), q0 = w, q1 = x, q2 = y, q3 = z

According to the [glm documentation](http://glm.g-truc.net/0.9.6/api/a00093_source.html) (line 102), glm::quat expects w, x, y, z.

So why is it initialized with z, y, x, w?

thp, 2016-12-04:
@cboulay  I've just tested this with `test_opengl` and this was indeed wrong (it worked before, possibly two errors cancelling each other out, and with the new orientation code, the single error had nothing to cancel out..).

Fix in #279

commit 185086c0 (2016-12-04): Merge pull request #279 from thp/fix-modelview-matrix

psmove_fusion_get_modelview_matrix: Fix orientation (Fixes #157)
commit fdb95b1c (2016-12-04): psmove_fusion_get_modelview_matrix: Fix orientation (Fixes #157)
closed 2016-12-04 by thp