thp/psmoveapi #172: psmovepair.exe has stopped working.

PolGraphic, 2015-05-05:
Hi.

I run the psmovepair (on laptop, windows 8, x64) and get by default:
`Connected controllers: 0`

When I plug the PS Move via usb I get:
`Connected controllers: 1`

but immediately after that, I get `"psmovepair.exe has stopped working"` window and the program terminates.

cboulay, 2015-05-05:
Windows 8.0 or 8.1? If 8.0, please try upgrading (it's free and better).
If 8.1, then I'd like to work with you to solve this problem. Answer back and I will compile a version of psmovepair.exe (and the dll's) that print out more information. I think there might be a couple different versions of the psmove controllers that have different behaviour in Windows.

By the way, in Windows 8, psmovepair by itself is not sufficient to establish a connection. You will also need a modified version of psmove-pair-win. I have a compiled version [here](https://github.com/cboulay/psmove-ue4/tree/master/Binaries/Win64). You can read more on the Wiki [here](https://github.com/cboulay/psmove-ue4/wiki).

PolGraphic, 2015-05-05:
Thank you for your attention.

I've the latest Windows 8.1.

When I run the `https://github.com/cboulay/psmove-ue4/blob/master/Binaries/Win64/psmovepair.exe` I got:

[image]

Which looks like some progress ;)

I've followed the instructions in application, but I get this error each time I try to press the "PS" button on PS Move.

cboulay, 2015-05-05:
Yes that's normal. Now you have to run psmove-pair-win.exe from that same folder.

PolGraphic, 2015-05-05:
Ok.

It initially displayed:
[image]

But then It just looping throw the same messages:
[image]

and PS Move stays blinking to me.

The `Successfully connected device" message is prommising, but the `ERROR ... Failed to get registry value: More data is avaiable." does not look too nice ;)

cboulay, 2015-05-05:
Can you check your registry (regedit.exe) at that path to see if it exists and if VirtuallyCabled is set to 1? If that path does exist but VirtuallyCabled is set to 0, then can you try to set it manually? (And I'll ask you to report this in the [psmove-pair-win repo](https://github.com/nitsch/psmove-pair-win))

If the path does not exist then try using [this method](https://code.google.com/p/moveframework/wiki/PairingConnectingInstructions), but skip the first part (setting the host address) as that was done by psmovepair.exe, and skip any mention of MotionInJoy.

PolGraphic, 2015-05-05:
The path does exists and `VirtuallyCabled` was set to `0`. I've changed it manually to `1` and reported the problem in the repo you've mentioned: https://github.com/nitsch/psmove-pair-win/issues/4

I now have something like this with `psmove-pair-win.exe`:
[image]

During the execution my PS Move turned off (I guess because of the battery, I will re-charge it and check again later). 

But, can I say that the device has been paired successfully? Or there still something to do now?

I've tried to run the `magnetometer_calibration.exe` but it just quits as soon as it starts.

cboulay, 2015-05-05:
After you made the manual change to the registry, it should not have been necessary to run psmove-pair-win. You should just be able to connect by pressing the PS button on the controller.

Most of the programs will quit if there are no controllers connected. If you run the program from a command prompt then you can see any messages it outputs.

As for this issue, I don't know why psmovepair.exe from this repo crashes but the one I provided you works. The one I provided you is from a branch in my fork of the psmoveapi, but there are quite a few things that are different so it'll be tough to nail down the exact reason... and I don't have time to do that right now.

PolGraphic, 2015-05-05:
I see.

I've press the PS button on the controller and it keeps blinking. Shouldn't it stay idle (with the color)?

The `magnetometer_calibration.exe` fails silently, even in command prompt (as administrator).

cboulay, 2015-05-05:
oh right. Sorry, the magnetometer_calibration does not print out how many controllers it found, while test_tracker does.

Yes the LED should stay solid.

Have you restarted the computer (maybe just restart the BT service) since editing the registry?
What bluetooth dongle are you using? Do you have any others you can try? Can you try plugging them into different USB ports?

I use an ASUS-BT400 and have never had any problems. Others report Belkin dongles work well for them. You're not the first to report the problems you're having, and in every case it is not an ASUS or Belkin dongle.

PolGraphic, 2015-05-05:
I've restarted the laptop.
Then press the PS button on controller (it blinks for minutes now).

The `psmove-pair-win.exe` shows:
[image]

But the register is (I've edited it under `H_KEY_LOCAL_MACHINE`, that's correct?):
[image]

When I run `test_tracker.exe` I get:
[image]

I have `Qualcomm Atheros AR3012 Bluetooth 4.0`:
[image]

cboulay, 2015-05-05:
I didn't expect test_tracker to work, unless you already setup the PS Eye camera (see the wiki link above). But it does report how many controllers it found. Was the controller plugged in by USB when you ran test_tracker?

Now that the controller is 'installed' and you have the correct registry entry, you should never have to run psmove-pair-win again. You should be able to connect simply by pressing the PS button on the controller. It should blink quickly while attempting to connect, then go solid once it is connected.

If it is plugged in by USB then it will continue to blink slowly, indicating that it is charging.

PolGraphic, 2015-05-05:
The controller was not plugged by usb. But the LED does not blink or even turns on anymore when I press the PS button :/ And for sure, it does not stay still (on). So, when I press the PS button, there is simply no reaction on LED at all (it stays dead). 

However, the `test_tracker.exe` says `found 1 controllers` (the LED is off). I've taken the controller to another room, behind two pairs of doors, and the program stills says about 1 controller.

When I plug the usb to charger, then press PS button, the icon is blinking (as you said, "charging").

I've not yet configured PS Eye.

cboulay, 2015-05-05:
Sometimes Windows does not correctly register a disconnect. When that happens I usually restart, but there's probably a way you can do it through "Devices and Printers", though you may end up uninstalling the controller (making it necessary to use psmove-pair-win and edit the registry again).

Try restarting. Before you press the PS button verify the test_tracker is reporting 0 controllers.
Then press the PS button, and hopefully it connects and stays connected. If it does not then see how many controllers test_tracker reports. If it's reporting that a controller is connected then there's a problem, and my first guess is that it would be an incompatibility with the Bluetooth Dongle. My second guess is that there is a variant of the PSMove controller that has a slightly different interface.

Without the problematic hardware, I cannot reproduce the problem and fix it.

PolGraphic, 2015-05-05:
Once again, thank you for the assistance :)

I've restarted the laptop, but it says `1 controller` (the PS Move lays near the laptop, I am charging it via usb to charger now).

I've turned on Airplane mode and turned it off (to restart the wifi/bluetooth fast).

I got `0 controllers`.

Then I pressed the PS button. The LED is blinking all the time (not solid :/ ). But the charger is still plugged in. I got `1 controller` :)

However, I cannot get the solid LED icon. Maybe it's because the PS Move is simply discharged? When I unplug it from the charger, the PS button does not turn on the LED. When It's plugged, the PS button cause the LED to blink all the time. I cannot get the still (on) LED either way.

I will update my post after some charging (I hope, that's the problem).

PolGraphic, 2015-05-05:
I guess I've finally got it working :D

The `magnetometer_calibration` turns on the bulb in PS Move via bluetooth :)
However, the LED stays blinking (weird, but working).

Unfortunately, the UniMove plug-in for Unity does not work (because of some x32/x64 conflicts with `libpsmoveapi.dll`. I've not found the x64 version of that dll :/

I guess I could write my own plug-in, but could you maybe point me how to get the PS Move orientation by myself in e.g. c++?

PolGraphic, 2015-05-05:
I guess I've finally got it working :D

The `magnetometer_calibration` turns on the bulb in PS Move via bluetooth and somehow reacts to its orientation (displaying some number) :)
However, the LED stays blinking (weird, but working).

Unfortunately, the UniMove plug-in for Unity does not work (because of some x32/x64 conflicts with `libpsmoveapi.dll`. I've not found the x64 version of that dll :/

I guess I could write my own plug-in...
Is there any console program that returns the orientation of PS Move (I could use it's output for the plug-in - I know it's not the best way, but it would be simple at least)?

cboulay, 2015-05-05:
Make sure the controller is not plugged in by USB when you are testing it. I think you can still do magnetometer_calibration when it is plugged in by USB, but I think the values it gives you are garbage so it's not really a good calibration. I'm not sure.

The dll's in my UE4 plugin repository that I linked above are 64-bit.

I've never used UniMove, but maybe UniMove in Windows expects 32-bit binaries because it expects use of the CL-Eye driver (the paid closed-source driver that is 32-bit only). Before I modified the PSEYEDriver to work in Windows, 32- and 64-bit, the CL-Eye driver was the only way to use the PSEye in Windows.

If you setup the PSEye following my instructions in the Wiki above and if you build psmoveapi for 32-bit following the directions to also use local opencv, and local libusb, then it you'll get 32-bit binaries that work without the CL Eye driver. However, I have no idea how to get this to work with UniMove. Maybe it'll just work.

PolGraphic, 2015-05-06:
If your dlls are 64-bit then that's great :)

Where can I find which functions do the `psmoveapi.dll` and `psmoveapi_tracker.dll` provides (so I can call them from c++ application)? I need the orientation and position of PS Move (the buttons would be nice, but they are not required).

cboulay, 2015-05-06:
The include files can be found [here](https://github.com/cboulay/psmove-ue4/tree/master/ThirdParty/psmoveapi/include). You can take a look at [these lines](https://github.com/cboulay/psmove-ue4/blob/master/Source/PSMove/Private/FPSMoveWorker.cpp#L67-L195) as an example of how to use the psmoveapi, including position, tracker, buttons, and rumble.

PolGraphic, 2015-05-06:
Thanks :) You're great.

Now I finally got it all working together :D

PolGraphic, 2015-05-06:
One more problem (which is similar to the earlier one, with registry).

I've installed the PS Eye (https://github.com/cboulay/psmove-ue4/wiki/Windows-PSEye-Setup, I can see it in device manager and the `ps3_eye_sdl.exe` works fine).

But when I try to run e.g. `test_tracker.exe` fails with the error:
[image]

I do not have such entry in registry.

cboulay, 2015-05-06:
Are you sure you're using my dll's? I removed the need for the registry entries for PSEye in my new_tracker branch. See [here](https://github.com/cboulay/psmoveapi/commit/e70330b9b0c7b064dc90a16fb8d61cd01d9cb47d).

PolGraphic, 2015-05-06:
Ok, I was using the psmoveapi custom build.

But the thing is, the demo works with... my small laptop built-in camera instead of PS Move camera :p Can I change it somehow? The PS Move camera is installed and connected.

cboulay, 2015-05-06:
Yeah it gets a little confusing because a lot of my changes haven't made it upstream yet. I've mostly been working on Windows-specific fixes and the upstream maintainers don't test in Windows, and even I only use Windows about once per week. And, in the months that I've been working on this, some conflicts have sprung up. It'll take me a significant amount of time to rebase off the upstream repo, with no guarantee that anything will be merged.

The upstream repo had some legacy code in it specific to Windows + PSEye that stored the PSEye settings into the registry upon camera opening, then restored them from the registry upon camera closing. The linux version does the same thing but it uses a file in the home directory. I changed the Windows version to do the same, no longer needing a registry entry. This only affects Windows + PSEye. Any other camera won't be affected. Any other OS won't be affected.

So you can either use my dll's, build my new_tracker branch, or use [this](https://raw.githubusercontent.com/cboulay/psmove-ue4/969df2ad147ed700da7626f868fe3a9b24f21742/Binaries/Win64/ps3eye_settings.reg) file to manually enter the expected registry settings.

I don't really want to use this issue to discuss this topic, but if you use my new_tracker branch then you should know it uses an entirely different [tracker algorithm](https://github.com/cboulay/psmove-ue4/wiki/Tracker-Algorithm). Others have reported it as being more accurate and I've measured it as being faster, but it's missing some smoothing and quality checks that thp has in his tracker.

PolGraphic, 2015-05-06:
Ok. Let's end the discussion. Thank you once again for solving the problem and for great support :+1:

closed 2015-05-25 by thp