thp/psmoveapi #173: Cannot enable orientation (I've called `psmove_enable_orientation` and calibration data is available).

PolGraphic, 2015-05-06:
I build the `psmoveapi` from sources. I want to turn on the orientation functionality.

`psmove_has_orientation` returned false, so I've used `psmove_enable_orientation` (the `psmove_has_calibration` returns true). Unfortunatelly, `psmove_has_orientation` still returns false.

According to http://o.thp.io/tmp/psmoveapi-builds/2.0.0~2012-11-05+1e6ac178/psmove_8h.html#a15227b9a2714aaf355dc8082430c2b43 :

    The orientation tracking feature depends on the availability of an orientation tracking algorithm (usually built-in) 
    and the calibration data availability (as determined by psmove_has_calibration()). 
    In addition to that (...), you have to enable the orientation tracking manually via psmove_enable_orientation()).

I've done 2 out of 3 (I've called `psmove_enable_orientation` and calibration data is available).

So I guess that "the orientation tracking algorithm (built-in) is not avaiable".

What can I do with it? It's possible to make that algorithm available by building the psmoveapi with some other options?

I don't have PS Eye connected, but I guess it's not required for that functionality.

cboulay, 2015-05-06:
You have to enable [PSMOVE_USE_MADGWICK_AHRS via cmake](https://github.com/thp/psmoveapi/blob/master/CMakeLists.txt#L63).

thp, 2015-05-06:
As @cboulay said, enable it in cmake. Note that this makes PS Move API fall under the terms of the GNU General Public License, so this might or might not be acceptable for your use case.

PolGraphic, 2015-05-06:
Thank you :) I'm still amazed of how fast and accurate you react on this repository :)

And the license note is indeed worth an consideration (I did not think about it).

thp, 2015-05-25:
Closing as fixed.

alejost848, 2016-05-26:
Now that the [references to the old Madgwick AHRS implementation are removed](https://github.com/thp/psmoveapi/commit/6ff65d3007e19bc7650dd89927dddc2900122de3) how can I enable orientation? I'm getting this warning `[PSMOVE WARNING] Built without Madgwick AHRS - no orientation support` when running the orientation sketch in Processing.

thp, 2016-05-28:
If you get this warning, it means that you haven't built the latest source version of PS Move API. Check out the code in the master branch, orientation tracking is now unconditionally enabled (the message you are seeing doesn't appear in the code anymore, so the build you have is based on some old code).

thp, 2016-05-28:
Or to be more precise, you still need to call `psmove_enable_orientation()`, but the build-time conditional has been removed.

alejost848, 2016-05-29:
I had already built the latest version from the repo and after some time of scratching my head I realised I forgot to update the library in the Processing libraries folder. Updating the psmoveapi.jar did the trick, thanks.

thp, 2016-05-29:
Marking as fixed then, thanks :)

closed 2016-05-29 by thp