thp/psmoveapi #183: auto dimming seems broken

cboulay, 2015-07-16:
[This chunk of code](https://github.com/thp/psmoveapi/blob/master/src/tracker/psmove_tracker.c#L957-L968) suggests to me that you are checking to see if the average colour in the illuminated sphere has a saturation of > 128, then stop, we're good with this dimming. However, if the saturation is less, then make the sphere dimmer! What!? Am I missing something?

Also, what's the point anyway? Why not always use maximum brightness? Under what circumstances would making the sphere less-bright give you better tracking?

If it's meant to give you better battery life, then wouldn't you want to start at the lower range of dimming then increase until you crossed 128?

P.S. While you're looking at this section of code, I think [line 936](https://github.com/thp/psmoveapi/blob/master/src/tracker/psmove_tracker.c#L936) is supposed to have `mask` here instead of `diffs[0]`.

cboulay, 2015-07-17:
OK I think I get it now. I just tried in a REALLY dark room and I saw what was happening. When the room is dark, and exposure is high, the sphere appears white at full brightness. So then it dims the sphere until it appears more colorful (higher saturation) and less white.

But I have the opposite problem in well-lit rooms. With the exposure low, as it needs to be in a well-lit room, the sphere may still be just-shy of the needed saturation value, so it starts to dim. However, in a well-lit room, any further dimming brings the sphere closer to white -> less saturation. So the sphere continues to dim until it's off, then the tracker doesn't work.

Why is it important that the sphere have high saturation? My guess is that the colour filters work better when it has high saturation, but I haven't tested this.

I think a short-term fix could be to check to see if the new saturation is larger or smaller after dimming. If smaller, then increase brightness, if larger then decrease brightness, and only continue in that same direction up to either limit.

But probably a better long-term solution will be to come up with a calibration method that searches multiple parameters simultaneously, including base LED colour. For example, I just tested a few different colours and, in this room, blue had very high saturation, and magenta was the worst.

Is it possible to guess at which colour might work best based on the LED-off image? Maybe we can sort the colours? The first will be the one with a hue furthest from the avg hue of the LED-off image, and the following will be the one with a greatest total distance from all previously selected colours?

cboulay, 2015-07-17:
(Sorry for using the issue tracker as my personal idea pad, but I'd like any input)
How about the following?

1. Find the exposure that maximizes saturation while minimizing luminance with the LED off.
    * Turn LED off
    * Binary search for exposure that minimizes alpha*luminance - beta*saturation (alpha and beta are TBD constants)
2. Sort the available LED colours
    * First colour will be chosen based on distance to LED-off hue at optimal exposure
    * Remaining colours will be chosen based on max-squared-difference to all preceding colours
3. For each controller, do the blinking calibration to maximize saturation:
    * Start at full brightness
    * If the sat isn't above threshold, try dimming more
    * if dimming reduced sat, go to previous dimming and stop

NoxWings, 2015-08-25:
You are right about the light dimming, I ran into problems while implementing psmove support into FreePie a while ago so I finally ended up turning off this feature.

thp, 2016-05-17:
The initial thing with reducing brightness was in cases where the sphere when too brightly lit would appear as white in the camera image, and reducing the brightness would lead to the sphere eventually appearing colored. This must have been with dark rooms where the exposure was set very high (to lighten up the surroundings) which caused the controller to light at full intensity.

Closing this in favor of #208, let's continue the discussion there.

closed 2016-05-17 by thp