thp/psmoveapi #185: Navigation Controller support status (OSX)?

RodrigoCard, 2015-09-10:
Hello!

I am interested in supporting the Navigation Controller in my Mac OS X Application.
By connecting through USB it detects as a regular joystick device but does not react to any input (I am using SDL2) See:
[image]
So, it is detecting as a proper HID joystick, I think...
Maybe it is a matter of just creating a simple driver to send the correct data,is it that simple?

I plan to do it on my own, or help get it to work, if I cant find a working solution, but I couldnt find enough useful info from my searches. So maybe this is a good place to ask?

Also, I found a reverted commit here with navigation support
https://github.com/thp/psmoveapi/pull/118
Was that working well? Wirelessly?

Thank you and sorry if I was confusing, I am a bit sleepy,

nitsch, 2015-09-11:
I have no idea if that part of the pull request was working or not.

In my own experiments (only on Windows and Linux), the problem with the
Navigation was always to reliably get it to connect via Bluetooth.
Actually communicating with the controller was never really a problem.

For some other info regarding the Navigation see the following
documentation:

https://github.com/nitsch/moveonpc/wiki/Navigation-Input-report
https://github.com/nitsch/moveonpc/wiki/HID-reports#user-content-Move_Navigation_controller

thp, 2015-09-21:
Navigation controller is basically like Sixaxis / DS3 in terms of connection. So you can use sixpair and all the Sixaxis-related utilities to make it work. Note that just connectinv via USB is not enough (see http://ps3.jim.sh/sixaxis/usb/ -- "However, the controller will still not report any events unless the mode is changed to "operational" using [a specific HID_REQ_GET_REPORT command](http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=4a1a4d8b87389e35c3af04c0d0a95f6a0391b964).").

nitsch, 2015-09-21:
> Navigation controller is basically like Sixaxis / DS3 in terms of
> connection. So you can use sixpair and all the Sixaxis-related
> utilities to make it work.

Yes, in theory. Never got the Sixaxis to work on my machine though using
these tools. Always needed other ways to work around that.

TrollinTeemo, 2016-07-18:
Is the psmove navigation controller now supported? I wanted to use it's accel data but all standard ps3 dualshock programs dont retrieve it.

thp, 2016-07-18:
The navigation controller doesn't have any sensors, just buttons and the analog stick.

nobodywasishere, 2018-03-30:
Any update on the status of this? The PS Nav controller is a fantastic one handed controller.

nitsch, 2018-04-01:
No PS Nav support has been added to the library yet, if that is what you
are asking.

thp, 2020-03-11:
There is now a PR with navigation controller support, see #403. On Linux, all features are supported, on macOS, most features (with the exception of analog trigger reading) are supported via Bluetooth. There's Bluetooth pairing support for both Linux and macOS via USB. Untested on Windows.

thp, 2022-12-17:
Closing as #403 is now merged.

closed 2022-12-17 by thp