thp/psmoveapi #187: Error connecting to PSMove #1 (Windows 10 64 bits)

M1st3rTea, 2015-10-10:
I have this error under Windows 10 64 bits :

Connected controllers: 1
Error connecting to PSMove #1

I tried to compile the api myself but the same problem occurs...

Any idea ??

nitsch, 2015-10-13:
> I have this error under Windows 10 64 bits :
> 
> Connected controllers: 1
> Error connecting to PSMove #1

Somebody else has already reported problems on Windows 10:

  https://groups.google.com/forum/#!topic/psmove/X5r5RC8XaZ4

Please have a look and let us know if the suggestions worked for you.

M1st3rTea, 2015-10-13:
I have troubles compiling libusb :/

nitsch, 2015-10-13:
> I have troubles compiling libusb :/

Not exactly sure why you would want to do that.

Have you tried with hidapi 0.7.0 as suggested? Have you tried picking up
debugging where Brett stopped (investigating the hid_open_path function
for instance)?

It is pretty hard for us to help since almost nobody here uses Windows 10.

M1st3rTea, 2015-10-13:
Well I thought, I had to use this fork https://github.com/cboulay/psmove-ue4
which needs libusb to be compiled.

Anyway this issue regarding libusb has been resolved by modifying C:\Program Files (x86)\Windows Kits\10\Include\10.0.10150.0\ucrt\time.h

> 
I had to comment these lines...
#ifndef _CRT_NO_TIME_T
    struct timespec
    {
        time_t tv_sec;  // Seconds - >= 0
        long   tv_nsec; // Nanoseconds - [0, 999999999]
    };
#endif

So which fork do you suggest I use ?

cboulay, 2015-10-13:
You linked to my UE4 plugin. It's not a fork of psmoveapi, but it uses a compiled version of my psmoveapi fork.

And you really don't want to modify your Windows system libraries.

M1st3rTea, 2015-10-13:
Oh my, it's getting late, well I only modified it in order to try to compile but in the end it failed.

thp, 2015-10-14:
So, should we move this issue report over there and close it here for now (at least until we merge the changes from there into this repo)?

M1st3rTea, 2015-10-14:
Well the original problem is in both versions.

nitsch, 2015-10-14:
> Alright, despite having done some useless stuff before, using hidapi 
> 0.70 helped with the the connection error.

So, the downgrade really made a difference? Interesting ... Maybe you
could investigate a bit further and find out which commit exactly in
hidapi breaks it for you on Windows 10 and report back. That would be
really helpful.

> Also, using the latest git version didn't help.

In case you are still talking about hidapi: Yes, versions more recent
than the exact version configured in psmoveapi's hidapi submodule are
known to break on Windows.

> Now I only have to press the playstation button to connect :D

That is how it is supposed to work once you could convince Windows to
make the Bluetooth connection.

> Now let's say I want to use the api to control the mouse in a game, 
> would there be anything to do it ?

If you are talking about arbitrary games or other applications (i.e.
nothing you have the sources to): There is nothing in psmoveapi to
directly map the controller to some generic input device interface on
Windows, as far as I know. This is actually what the HID interface is
for. But the Move controller is not really playing by the rules. Hence
the need for things like psmoveapi to really use the controller's full
potential (reading its sensor data, using the camera for tracking). The
downside is that you have to take care of a lot of stuff yourself.

But I did not really investigate generic mouse input and the likes on
Windows. So I am not the best guy to answer this question. ;)

> Also, can I shutdown the psmove once it's on ?

Not really sure what you mean.

In Windows versions < 10, you can select the Move from the list of
Bluetooth devices and disconnect it. I suppose that still works in
Windows 10.

M1st3rTea, 2015-10-14:
I wanted to know about powering off the psmove. Well I simply have to keep pressing playstation.

I will identify the latest working hidapi.

nitsch, 2015-10-15:
> I wanted to know about powering off the psmove. Well I simply have to keep
> pressing playstation.

Yes, that will work, too. But at least for Windows version < 10 there was
a problem with that approach. Windows did not properly recognize the
disconnect and screw up the next time you connect something and run
psmoveapi:

  https://groups.google.com/forum/#!topic/psmove/njN8hoIaWKM

Does that still happen on Windows 10?

jpablolastra, 2015-10-19:
Hey, thanks for all the info and pointers. I had the same issue that started this thread (win 10 + error connecting to ps move #1) and I can confirm than moving to hidapi 0.7 also fixed it.

Unfortunately, even though I pass that step, I keep getting the following message on psmovepair, no matter how many times I press the PS button when the light turns off.

    D:\Development\psmoveapi\build>psmovepair
    Connected controllers: 1
    [PSMOVE WARNING] Magnetometer in 04:76:6e:00:6b:64 not yet calibrated.
    PSMove #1 connected via USB.

    Unplug the controller.

    Now press the controller's PS button. The red status LED
    will start blinking. Whenever it goes off, press the
    PS button again. Repeat this until the status LED finally
    remains lit. Press Ctrl+C to cancel anytime.

    A connection could not be established. This is not
    unusual in Windows. Please refer to the README document
    for your platform for more information. Press Ctrl+C to cancel.
    A connection could not be established. This is not
    unusual in Windows. Please refer to the README document
    for your platform for more information. Press Ctrl+C to cancel.
    A connection could not be established. This is not
    unusual in Windows. Please refer to the README document
    for your platform for more information. Press Ctrl+C to cancel.
    A connection could not be established. This is not
    unusual in Windows. Please refer to the README document
    for your platform for more information. Press Ctrl+C to cancel.

Any other trick that could help me pass that setup step?

Thanks!

jpablolastra, 2015-10-19:
Something that may be worth mentioning. If I check the bluetooth connections panel in Win10, it shows a "Motion Controller" and sometimes it says "pairing", sometimes it says "paired", and sometimes it even says "Connected", even though the red light is either off or blinking. There seems to be no relation between the light and the status that win10 shows while psmovepair is running.

If I cancel psmovepair, then the win10 status for the bluetooth connection keeps as "paired". I tried to "Remove device" in that same window, but it throws an error and the device stays there.

Seems that I'll need to rollback to Win7 :(

cboulay, 2015-10-19:
@jpablolastra It seems none of the README's have an important piece of information for Windows > 8.
What you're seeing is totally expected. There's an additional registry hack necessary for Win 8 (and I guess Win 10). @nitsch made a utility to do that hack for you, found [here](https://github.com/nitsch/psmove-pair-win). Or you can download the binary [here](https://github.com/cboulay/psmove-ue4/blob/master/Binaries/Win64/psmove-pair-win.exe) (click "view raw" to start the download).

Someone rolled that step into psmovepair but it sometimes leads to BSOD, whereas in separate steps it never does, so they remain separate for now.

M1st3rTea, 2015-10-19:
well, to pair I let the psmovepair do its fail job and I ran psmovepair (or psmove-pair-win) from cboulay's repo for his UE4 implementation https://github.com/cboulay/psmove-ue4
I really didn't investigate if it helped in any way but I had both opened and I keeped pushing the playstation button without usb.
Eventually it worked fine ;)

Try it out and tell me .

jherico, 2015-10-20:
I'm seeing the same issue.  I was able to get past the initial problem by downgrading HID to 0.7 as suggested.  I ran a git bisect to identify the problem commit in hidapi and it reported  e3ab121099376d489a44e16bc62e2c035bfcaa9f was the commit that broke it.

jpablolastra, 2015-10-21:
Running psmove-pair-win (and having a bit of patience) did the trick.
Thanks a lot guys!

nitsch, 2016-02-13:
Please note that using `psmove-pair-win` is no longer necessary. I have
integrated its functionality into psmoveapi's `psmovepair.exe` utility and
it also works for Windows 8 and 10.

thp, 2017-09-10:
This should be fixed with #319 (for the hidapi fix) and #318 (for running the tool as Administrator).

closed 2017-09-10 by thp