thp/psmoveapi #188: How to run with CL Eye Driver

omerfaruk, 2015-10-29:
Hi,
I have cl eye driver.I paired my move with bluetooth. I connected my camera and move.When i start app camera not working and move sphere not have color.How can i run with cl eye driver ?

My system: Win 10 x64

cboulay, 2015-10-30:
Win 10 doesn't work with psmoveapi in its current form because of a problem with hidapi. You have a couple options. You can either roll back the hidapi to an older version (<= 0.7) or you can use a newer hidapi + modify hidapi + modify psmoveapi to work with the new hidapi.

I've taken the latter option in [my psmoveapi fork](https://github.com/cboulay/psmoveapi) and it works with both Win 10 and OSX. But, in both those platforms, I use the PS3EYEDriver for communicating with the camera and I haven't tested CL Eye. It may or may not work because it may or may not know where to find the CL Eye includes and libs.

cboulay, 2015-10-30:
[Here](https://github.com/cboulay/hidapi/commit/f765e37269f60be1644b6beeae2ede0514798c1c) is the tiny modification to hidapi to get it to work with Win10. This might be all that's necessary for the upstream psmoveapi to work.

In my fork I made additional changes to psmoveapi because my hidapi is forked off the latest, and a while back hidapi introduced yet another problem where feature reports are 1-byte longer in Windows than they are in Mac. I worked around this so my psmoveapi is using the newest hidapi with 1 slight modification.

Ps2Fino, 2016-01-22:
Hi @cboulay Does your fork work on OSX 10.10.5 or a Raspberry PI?
I'm trying to get the tracker application working with OpenCV 3

cboulay, 2016-01-22:
I use it in OSX 10.11 all the time so it should work in 10.10. I don't think it will work with OpenCV 3 though. I think psmoveapi uses some deprecated includes or functions... I can't remember the details.

thp, 2016-01-27:
I wonder if we should drop support for the CL Eye Driver in favor of also recommending PS3EYEDriver on Windows as well?

cboulay, 2016-01-27:
Unfortunately, I think it's worthwhile to keep CL Eye Driver support around.

PS3EYEDriver performance is significantly worse than CL Eye driver. libusb for Windows does not perform very well, and that is compounded by PS3EYEDriver being a little inefficient in how it makes libusb calls. CL Eye Driver, on the other hand, uses native Windows APIs and performs quite well.

@rovarma has some PS3EYEDriver changes that improve performance quite a bit in Windows but he's still working on the cross-platform compatibility. Even then, I doubt PS3EYEDriver will quite reach the performance of CL Eye driver, but it may be good enough.

Another issues is that PS3EYEDriver is GPL'd. There are psmoveapi users in the wild that use CL Eye driver for exactly this reason.

rovarma, 2016-01-28:
As @cboulay said, I don't think we can drop support for CLEyeDriver completely (because of the GPL virus), but I would personally be in favor of making PS3EYEDriver the "official" driver on Windows. 

Performance wise, I'm sure we'll be able to match the CLEyeDriver with the changes I'm making. The whole reason I started looking into improving PS3EYEDriver in the first place is to avoid using CLEyeDriver. I'm highly allergic to companies/people charging money for products that are not supported anymore.

nitsch, 2016-02-01:
What about dropping support for the CLEye SDK? I have not tried
recently, so is there any reason for keeping this?

thp, 2016-02-01:
Yeah, CL Eye SDK support should be dropped IMHO. That's different from the CL Eye Driver.

Ps2Fino, 2016-02-02:
Out of interest, what are people's experiences of PS3EYEDriver on Linux? I built the PSMove API on my Ubuntu box at home and the tracker works sometimes and sometimes not.

Not sure if this is a tracker issue or a driver issue (or indeed that my machine is a 2.2 dual core from 2009.....)

rovarma, 2016-02-02:
I only work on Windows, but I imagine the PS3EYEDriver should work fine on Linux. It uses libusb to communicate with the camera, which is cross platform.

I've recently been working on some performance improvements to the driver, which have been accepted in this PR: https://github.com/thp/PS3EYEDriver/pull/15. Note though that that version of PS3EYEDriver does not work with the current version of PSMoveAPI; that's pending this PR: https://github.com/thp/psmoveapi/pull/200.

nitsch, 2016-02-02:
Out of interest, why do you want/need to use PS3EYEDriver on Linux? The
PlayStation Eye is natively supported since ages (kernel version 2.6.29).

thp, 2016-05-17:
Pull request for removing CL Eye SDK support here: #235

This still allows for the possibility of using the CL Eye Driver on Windows.

thp, 2016-05-17:
Closing as fixed by removing the CL Eye SDK support.

closed 2016-05-17 by thp