thp/psmoveapi #189: Error bulid psmoveapi_tracker

czhzng, 2015-12-02:
I have this error under MSVC 2013.

fatal error C1083: Cannot open include file: 'ps3eyedriver.h': No such file or directory

where can i found it?

thanks!

thp, 2015-12-02:
Did you do a `git clone --recursive` or `git submodule init && git submodule update`?

czhzng, 2015-12-02:
yes, I do.
in project "psmoveapi_tracker", I defined CAMERA_CONTROL_USE_PS3EYE_DRIVER.

cboulay, 2015-12-04:
It was my fault. He/she was using my fork. It's too bad github doesn't allow issues on forks.

I had a dangling reference to a submodule that was preventing the --recursive or init && update from completing.

thp, 2015-12-06:
Ok, I guess I can close this as working now? :)

thp, 2015-12-06:
Closing. Please reopen if there's still something I can do to help here.

closed 2015-12-06 by thp