thp/psmoveapi #191: Compiling OpenGL examples on Win32

Ixstala, 2016-01-05:
I'm having trouble getting the GL examples to compile. Can you point me to the other dependencies required to compile them? Obviously it's SDL and some form of GL, but I'm not sure which versions and I don't know where they should live relative to mingw. This isn't in the build documentation.

Thanks!

thp, 2016-01-06:
SDL and OpenGL. It could be that you need to use GLEW to get it running on Windows (and it might entail some source code modifications). The OpenGL examples have mostly only been tested under Linux.

cboulay, 2016-01-06:
We build the opengl examples in Windows in my fork. The relevant cmake section is [here](https://github.com/cboulay/psmoveapi/blob/master/examples/CMakeLists.txt#L78-L108).

Ixstala, 2016-01-06:
I've got cboulay's fork building the examples just fine. Now that my 32bit build is no longer crashing I don't need to build the examples here.

Thanks anyway!

closed 2016-01-06 by Ixstala