thp/psmoveapi #192: Error with building package

BrownRecluse, 2016-01-13:
I've followed the README.ubuntu file and when I got to 
6.a Build the API (as Debian package) and install it
   cd ~/src/psmoveapi/
   dpkg-buildpackage -rfakeroot
   sudo dpkg -i ../*psmoveapi*.deb

When i enter dpkg-buildpackage -rfakeroot I get an error saying 
tail: cannot open ‘debian/changelog’ for reading: No such file or directory
dpkg-buildpackage: error: tail of debian/changelog gave error exit status 1

thp, 2016-01-15:
Oh yeah, in the recent builds we have removed the `debian/` folder as it was outdated and unmaintained. Try this instead:

```
mkdir build
cd build
cmake ..
make
make install
```

thp, 2016-02-12:
Closing as fixed - the README file was updated.

closed 2016-02-12 by thp