thp/psmoveapi #193: Windows7 x64 vs2013:PSMoveTrackerSmoothingSettings error C2078:too many initializers

bEenAccepted, 2016-01-18:
OS: Windows7 x64

Platform: VS2013

Error: error C2078:too many initializers

Source code:
// Smoothing defaults
static const PSMoveTrackerSmoothingSettings tracker_smoothing_default_settings = {
    .filter_do_2d_xy = 1,
    .filter_do_2d_r = 1,
    .filter_3d_type = Smoothing_None,
    .acceleration_variance = 10.f,
    .measurement_covariance = { 0.01f, 0,   0,
                                0,   0.01f, 0,
                                0,   0,   1.0f}
};

I don't know the meaning of "." , is this a new operator of c++11? The other project works fine, except the "psmoveapi_tracker".

cboulay, 2016-01-18:
This is in reference to my fork, not the upstream (original) repo where this issue is posted. The maintainers of this repo are probably pretty confused.

Maybe you can post the issue in https://github.com/cboulay/psmove-ue4 ? Sure, the error is specific to the psmoveapi and not the UE4 plugin, but unfortunately github doesn't let you post issues on forks.

To answer your question, it's [C99](https://gcc.gnu.org/onlinedocs/gcc/Designated-Inits.html).

thp, 2017-09-10:
Closing this issue, please file a new issue at https://github.com/cboulay/psmove-ue4 if this is still an issue there.

closed 2017-09-10 by thp