thp/psmoveapi #194: Automate building/fetching OpenCV and other dependencies

nitsch, 2016-01-19:
When using a locally built OpenCV, instead of relying on the user to build it, we could automate this part using CMake's [ExternalProject](https://cmake.org/cmake/help/v3.2/module/ExternalProject.html). Note that this also allows adding patches.

Might be useful for handling other dependencies in psmoveapi.

cboulay, 2016-01-19:
By 'locally built' you mean the OpenCV in the psmoveapi/external, correct?
I'm OK with this, as long as it's smart enough to recognize if it's already built then it shouldn't build it again. Whenever I'm developing cmake code, I like to wipe out my build directory and start over with a fresh cmake command, because I never know what's stored in the cmake cache and how that could be affecting my edited cmake scripts. This would be annoying if I also had to re-buidt OpenCV every time I did that.

nitsch, 2016-01-19:
> By 'locally built' you mean the OpenCV in the psmoveapi/external, 
> correct?

Yes.

> I'm OK with this, as long as it's smart enough to recognize if it's 
> already built then it shouldn't build it again.

It does.

> Whenever I'm developing cmake code, I like to wipe out my build
> directory and start over with a fresh cmake command, because I never
> know what's stored in the cmake cache and how that could be [...]

Well, then don't do that. ;)

We could still have a CMake option that overrides this behaviour an goes
searching for a compiled OpenCV on the system (or where ever you point it).

thp, 2016-01-27:
+1 for that; also, it could be useful in the CI once we get that set up.

thp, 2022-12-17:
This seems to be implemented for some time now. The build scripts in `scripts/` do build OpenCV from source.

closed 2022-12-17 by thp