thp/psmoveapi #195: “Building the API manually”

HUGEwindow, 2016-01-19:
Note: If this did not work, you might need to build the API step-by-step. See “Building the API manually” for the alternative procedure.

Where can i find this? 

I am trying to pair a move controller with macbook running el capetian 

Thanks in adavance.

nitsch, 2016-01-19:
It is in the very same document you are quoting.

HUGEwindow, 2016-01-19:
My apologies. 

This mean anything:
In file included from /Users/open/Dev/psmoveapi/src/platform/psmove_osxsupport.m:33:
In file included from /System/Library/Frameworks/IOBluetooth.framework/Headers/objc/IOBluetoothHostController.h:6:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSArray.h:6:
/System/Library/Frameworks/Foundation.framework/Headers/NSEnumerator.h:7:15: error: expected ';' after @class
@class NSArray<ObjectType>;

closed 2016-01-20 by HUGEwindow