thp/psmoveapi #196: PS Move and glow LED tracker

lazyspark, 2016-01-20:
Hi, 

I have got the example working fine on Win10. And as I was reading the documentation it says you can track upto 5 controllers. Instead if I want to use the tracker in the src to track more glow LEDs say 10 is it possible or there is any limitation on the program to be able to track only 5. Assuming that I don't pair remaining 5 units and they are just LED glow bulbs moving around and tracking something else (other than the controller)

Please let me know if the question is cryptic and thanks for creating this project. 

Sawan

nitsch, 2016-01-22:
Although I am not intimately familiar with every detail of the current
tracker implementation, I would say that tracking more than 5 blobs
should be possible. I believe the limit of 5 controllers comes mainly
from the Bluetooth communication -- which you do not need to track some
unrelated blobs.

You might need to put a little work into initially registering the blobs
(depending on their color) since the tracker assumes it can control the
glowing orb's color. But I believe the actual tracking should work with
your LEDs.

lazyspark, 2016-01-27:
Thanks for your response.

closed 2016-01-27 by lazyspark