thp/psmoveapi #197: PSMoveAPI test_tracker application crash

Ps2Fino, 2016-01-22:
Hi

I'm having an issue with the PSMove API tracking functionality on OSX.
Pairing controllers from USB, and interfacing with Bluetooth, and turning on/off LEDs/Rumble all work fine.
The issue is with the tracking API.

I've tried running the tracker application but it crashes after connecting to the iSight camera.
Here is a snapshot from the crash in Xcode: it's the call to cvCvtColor from OpenCV.

[image]

Any idea whats happening?

Cheers

Dan

cboulay, 2016-01-22:
In your other post you said you were trying to use OpenCV3. Is that the case here? Last I tried, psmoveapi does not work with OpenCV3.

Ps2Fino, 2016-01-22:
Yep, I have the latest build of OpenCV3, and having had a browse around this morning, I think you're right. I'll jump back down to OpenCV 2. Thanks for the reply! :-)

Ps2Fino, 2016-01-22:
Out of interest, I saw the Thomas has picked up development recently? Are there plans to update the API to OpenCV 3?

Ps2Fino, 2016-01-22:
@cboulay I've jumped down to OpenCV 2, yet I'm still getting the same issue
[image]

The assertion inside the implementation of cvCvtColor is failing: specifically it requires that dst.data == dst0.data, and it fails

dst0 is created from cvarrToMat function call, then dst is set to dst0.
Then a call to cv::cvtColor is called

cboulay, 2016-01-22:
I haven't tried the iSight camera in a very long time. It's possible something broke. Do you have a PS3Eye camera, just to test?

Ps2Fino, 2016-01-22:
I do, but not here in the office. I'll try it when I get home.
At this point, I'm almost certain it's an OpenCV issue as looking at the debugger in Xcode, all roi-s in the tracker struct have the same number of channels and channel sequence (BGR) as is expected by the call to cvCvtColor.
If I do track down the problem, I'll forward the solution here

nitsch, 2016-01-22:
> Last I tried, psmoveapi does not work with OpenCV3.

Good to know.

But why did you change the code in the Windows (MinGW) README from
explicitly using OpenCV 2.4.10 to using OpenCV's master branch then and
mentioning OpenCV 3, too? Especially since the MSVC README still uses
OpenCV's 2.x branch.

nitsch, 2016-01-22:
> Out of interest, I saw the Thomas has picked up development recently?
> Are there plans to update the API to OpenCV 3?

None that I heard of. But that is hard to tell reliably without some
official schedule. ;)

Feel free to tackle this transition to OpenCV 3 if it makes sense.
Should be a relatively small amount of changes necessary to do that.
Contributions are always greatly appreciated!

nitsch, 2016-01-22:
Hm, regarding OpenCV 3, this might be interesting information, even
though I did not find more recent statements regarding this issue:

http://answers.opencv.org/question/17546/opencv-will-drop-c-api-support-soon/

I never use OpenCV's C API, so I cannot comment on any real issues with
it. But since psmoveapi wants to stick to C, an update to OpenCV 3 might
actually introduce more problems than (possible) improvements. I guess
that really depends on the features used in the tracker.

cboulay, 2016-01-23:
> But why did you change the code in the Windows (MinGW) README from explicitly using OpenCV 2.4.10 to using OpenCV's master branch then and mentioning OpenCV 3, too? Especially since the MSVC README still uses OpenCV's 2.x branch.

I definitely got OpenCV3 pre-release working with psmoveapi in some build environment, but I can't remember the details. I've tried so many configurations, made so many changes that didn't make it upstream, or some that made it upstream in a different format, that I can't keep track anymore.

Ps2Fino, 2016-01-25:
@nitsch @cboulay Thanks for the comments folks! I haven't got time to dig into the root cause of my issues. For what its worth, I got hold of a PS3 Eye camera, recompiled the API with libusb, OpenCV 2, and using the eye camera for tracking. Everything seems to be working now, which makes me think that it is an issue with the iSight camera. 

Incidentally, stepping through the sample code in the debugger didn't cause a crash. The reason for the crash was that the camera frame (namely the IplImage struct) was being modified in between calls to the color space transform function cvCvtColor. My guess here is its a race issue with the iSight and general OSX thread concurrency shenanigans. Sticking with libusb and the eye camera for now :-)

Cheers

Dan

thp, 2016-01-27:
Ok, closing for now. The PS Eye is the recommended device, as we have an easy way of testing with it on all platforms, whereas the iSight does require specific hardware (the same applies to all other third party camera, it just so happens that the PS Eye is used with the PS Move also in "official" applications).

closed 2016-01-27 by thp