thp/psmoveapi #199: Inifinite loop in ps3_eye_grab_frame

Ps2Fino, 2016-01-29:
I've been having issues with the API for days now.
Initially, the API worked fine, and the test_tracker application worked and I could successfully track a controller.

However, I am now finding the ps3_eye_driver to be stuck in an infinite loop.
This seems like its more of a libusb/my PS Eye issue but wanted to discuss it in case anyone else is having the same problems.

I'm running OSX 10.10.5, and linking to openCV 2.4.10.2 tag, the PS3EyeDrive version tagged in the repo's submodule, and the libusb submodule also

[image]

cboulay, 2016-01-29:
You should probably wrap your print statement in yet another `if (!eye->eye_isNewFrame())`. As it is now, it'll print that statement even when it's not stuck in an infinite loop.

Ps2Fino, 2016-01-29:
Just have. Same issue

cboulay, 2016-01-29:
Well, you'll still get that print statement even during normal functioning, but it should be a little less frequent. Really what you should do is create a persistent variable (global is OK for testing I guess) that counts how many iterations you go through in that loop and you include that count in your print statement. Then, when you exit the while loop, set the counter back to 0.

Ps2Fino, 2016-01-29:
Thanks for the advice. Same issue sadly
[image]

Ps2Fino, 2016-01-29:
I'm going to pull the latest of the PS3EYEDriver repo, build that, then run the openframworks example. At least that way I'll be able to rule out the PSMove-API if the openframeworks fails also

cboulay, 2016-01-29:
IMO, the SDL example is a little easier to setup. But if you are already comfortable with openframeworks then that will probably be easier for you.

Let us know what happens.

Ps2Fino, 2016-01-29:
Yep! An issue with my camera perhaps!!!
Or libusb

[image]

Ps2Fino, 2016-01-29:
I've just tried plugging the camera into my laptop and running the SDL example from the PS3EYEDriver repo and I get a video stream. As libusb has been installed on both my machines via homebrew, this now leads me to assume the issue is a dodgy USB socket on my machine

Thanks for the help!
I'll update one more time after I have tried the PSMove-API on my laptop.
Of course, if Thomas wants to close this issue now instead, he's more than welcome to

cboulay, 2016-01-29:
I've had similar things happen to me in the past when I had a different usb device on the same USB controller. I now plug my PS3 Eye camera into a port on the front of my desktop and everything else goes in the back. I haven't had problems since.

Ps2Fino, 2016-01-29:
Yep: just annoying that I've lost a few days over something silly. Overthinking the problem all the time I guess. Thanks for all your help cboulay :-)

thp, 2016-05-17:
Closing the issue as per the last comment.

closed 2016-05-17 by thp