thp/psmoveapi #204: Build error under OSX El Capitan

joezavtra, 2016-02-16:
First of all, thank you guys for this lib!

I've used older version of psmoveapi from November i think. All was OK.
But after pulling new version it does not build.

Latest OpenCV from github was successfully built and installed.
But got following when I tried to build psmoveapi (cut out some parts):

**jz@mac:~/ws/psmoveapi/build [master]$** cmake ../
```
  Tracker
    Tracker library:  Yes
    AHRS algorithm:   None
    PS Eye support:   Yes
    HTML tracing:     No
    Use CL Eye SDK:   No
    Use PS3EYEDriver: No (disabled)

  Build configuration
    Debug build:      No
    Library license:  BSD (see README for details)
    BlueZ 5 support:  No (for pairing on Linux only)

  Additional targets
    C example apps:   Yes
    OpenGL examples:  Yes
    C test programs:  Yes
    C++ TUIO server:  Yes

  Language bindings
    Python:           No (SWIG not found)
    Java:             No (SWIG not found)
    C#:               No (SWIG not found)
    Processing:       No (SWIG not found)

-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   psmoveapi
   psmoveapi_tracker

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/jz/ws/psmoveapi/build
```

**Seems OK**

**jz@mac:~/ws/psmoveapi/build [master]$** make
```
Scanning dependencies of target psmoveapi
[  1%] Building C object CMakeFiles/psmoveapi.dir/src/psmove.c.o
/Users/jz/ws/psmoveapi/src/psmove.c:2165:45: warning: the value of the size argument in 'strncat' is too large,
      might lead to a buffer overflow [-Wstrncat-size]
        strncat(dir, PATH_SEP ".psmoveapi", sizeof(dir));
                                            ^~~~~~~~~~~
/Users/jz/ws/psmoveapi/src/psmove.c:2165:45: note: change the argument to be the free space in the destination
      buffer minus the terminating null byte
        strncat(dir, PATH_SEP ".psmoveapi", sizeof(dir));
                                            ^~~~~~~~~~~
                                            sizeof(dir) - strlen(dir) - 1
1 warning generated.
[  2%] Building C object CMakeFiles/psmoveapi.dir/src/psmove_calibration.c.o
#####
[... 323 lines cut ...]

[ 86%] Building C object CMakeFiles/tracker_camera_calibration.dir/examples/c/tracker_camera_calibration.c.o
In file included from /Users/jz/ws/psmoveapi/examples/c/tracker_camera_calibration.c:42:
In file included from /usr/local/include/opencv2/calib3d/calib3d.hpp:48:
In file included from /usr/local/include/opencv2/calib3d.hpp:47:
/usr/local/include/opencv2/core.hpp:49:4: error: core.hpp header must be compiled as C++
#  error core.hpp header must be compiled as C++
   ^
In file included from /Users/jz/ws/psmoveapi/examples/c/tracker_camera_calibration.c:42:
In file included from /usr/local/include/opencv2/calib3d/calib3d.hpp:48:
In file included from /usr/local/include/opencv2/calib3d.hpp:47:
In file included from /usr/local/include/opencv2/core.hpp:54:
/usr/local/include/opencv2/core/base.hpp:49:4: error: base.hpp header must be compiled as C++
#  error base.hpp header must be compiled as C++
   ^
/usr/local/include/opencv2/core/base.hpp:52:10: fatal error: 'climits' file not found
#include <climits>
         ^
3 errors generated.
make[2]: *** [CMakeFiles/tracker_camera_calibration.dir/examples/c/tracker_camera_calibration.c.o] Error 1
make[1]: *** [CMakeFiles/tracker_camera_calibration.dir/all] Error 2
make: *** [all] Error 2
```

Thanx

cboulay, 2016-02-16:
What options did you use to build and install opencv? Did you follow the directions [here](https://github.com/thp/psmoveapi/blob/master/contrib/build-osx-snapshot)?

I haven't tried in a while but those options definitely worked for me in the past.

Based on your error, it's not picking up opencv in the psmoveapi/external folder, rather cmake is picking up the system-installed opencv. Is that what you intended? If so then I wonder if maybe you didn't build all the opencv modules when you installed opencv? Can you try uninstalling the system opencv and instead installing it with homebrew?

cboulay, 2016-02-16:
I just noticed you did a `cmake ../`

I don't think you should run cmake with only the default options. I think the defaults might need fixing, but until then you should at least pass `-DPSMOVE_USE_PS3EYE_DRIVER=ON` for OS X if you're using the PSEye.

If you want to specify the folder in which to find OpenCV then also pass this option:
`-DOpenCV_DIR=<path_to OpenCVConfig.cmake>`

thp, 2016-05-17:
#236 merged, closing this as fixed. Please reopen if it's still an issue for you.

closed 2016-05-17 by thp