thp/psmoveapi #206: Too much yaw drift

cboulay, 2016-02-17:
@rovarma Has done a lot of good work to bring the upstream repo up to and past my fork in many regards. There are really only a few enhancements remaining in my fork that I think are worth brining up. I'm already declining to support my fork because I'm too busy but I would love to do that knowing that the upstream repo is feature complete.

The first feature is an improvement to the orientation code done by @brendanwalker . He rewrote the Madgwick AHRS code straight from the original paper, so it's no longer GPL'd. There are many other changes too that I won't list because the [code is available](https://github.com/cboulay/psmoveapi/blob/master/src/psmove_orientation.h).

The main benefit (despite the removal of GPL) is that the yaw drift is GREATLY reduced. The main drawback is that it's significantly more complex. It requires Eigen, is C++, and has a small custom math library in psmove/src/math. There's also an extra step added to the magnetometer calibration to initialize the orientation state, but it's pretty quick.

I don't have time to do this, but I think it's worthwhile if someone does have the time.

brendanwalker, 2016-02-18:
FWIW if including Eigen is too big of a change (which I can totally understand) I don't actually need to use Eigen for the new Madgwick-derived algorithm. I'm really just using that lib for it's general M*N matrix multiply and a quaternion class. Before I switched over to using Eigen (which we needed for HMD/coregistration stuff), I actually was using some super basic psmove C structs and associated functions for a vector, MxN matrix, and quaternion, which the orientation code used to use. If there was interest in bringing this change over I can make a less dependency heavy version of it and submit a pull request. You can see a comparison of the two filters here: https://www.youtube.com/watch?v=t4azU8rKym0.

thp, 2016-02-18:
Cool video! Yes, I'd be interested in getting this in, and removing the GPL dependency is indeed a good thing (it will make the CMake build system that tells which license the library falls under easier, and we can build in orientation tracking unconditionally) - assuming your code is under a BSD-like license like PS Move API is :)

As for the math dependency, maybe glm would work here (of course, then you need to have some C++ code wrapped away in extern "C" functions, but we already do this in [psmove_fusion.cpp](https://github.com/thp/psmoveapi/blob/master/src/tracker/psmove_fusion.cpp) and also use quaternion classes and matrices).

brendanwalker, 2016-02-18:
Yeah I put the same psmove api license over top of the new orientation code. I could use glm for most (all?) of the math. There is one 4x6 matrix multiplied with a 6x1 vector, but I think I can turn that into 2 (4x3)x(3x1) matrix/vector multiplications, which would allow me to use glm for there as well. If not I can just make special function for that and use raw arrays. In any case, I'll see if I can bang out that that change this weekend and get you a pull request to review.

rovarma, 2016-02-19:
Just wanted to say that it would be super cool to have this in this repo. That video looks like the new code substantially improves the orientation stability.

Let me know if I can help out in any way with getting this in.

BinaryShift, 2016-02-23:
@rovarma @thp @cboulay Pull request is up here: https://github.com/thp/psmoveapi/pull/215. Anyone feel like doing a test of this on their machine.  Even better if it's a non Win10 OS (OS X or linux).

rovarma, 2016-02-23:
Thanks for the efoort! That's excellent. 

I'll give it a shot tonight (Windows 8). Don't have a linux/osx box to test on, unfortunately.

thp, 2016-02-28:
Closing this one and we'll review/merge #215 soon-ish.

closed 2016-02-28 by thp