thp/psmoveapi #209: PS Eye setup on Windows

nitsch, 2016-02-22:
With the current setup and build config as described in docs/README.win64_msvc, what are the necessary steps to successfully run the `test_tracker` example? It seems there is some additional setup required to use the PS Eye camera since it does not seem to be recognized as connected inside `camera_control_new_with_settings()`.

nitsch, 2016-02-22:
It still seems to be necessary to manually install libusb drivers as described here: https://github.com/cboulay/psmove-ue4/wiki/Windows-PSEye-Setup.

We should have that description either in a PS Move API wiki or some other documents accompanying the sources.

Derzo, 2019-09-01:
How did you do it? I can't figure it out. What exactly am I supposed to do with PS3EYEDriver source?

cboulay, 2019-09-01:
Better community-contributed instructions for setting up drivers on Windows are here: https://github.com/psmoveservice/PSMoveService/wiki/PSEye-Software-Setup-%28Windows%29

Derzo, 2019-09-01:
Thanks, I'll check it out.

thp, 2022-09-17:
This basically boils down to installing Zadig on Windows and then using the libusb-based PS3EYEDriver, right?

thp, 2022-12-20:
Probably implemented by https://github.com/thp/psmoveapi/pull/459/commits/5a21b2a0ab4994eaa04bed9a9f511a01e7a07227 in #459 and will be fixed once that's merged.

closed 2022-12-23 by thp