thp/psmoveapi #214: Fix hidapi on Windows 10

nitsch, 2016-02-22:
hidapi needs to be adapted for Windows 10. See https://github.com/thp/psmoveapi/issues/187#issuecomment-149458573. Will the suggested dowgrade work on other platforms (including older versions of Windows) as well or are we losing something else in the process? Should Windows 10 simply user a different version of hidapi while everybody else sticks to the current one?

cboulay, 2016-02-22:
[I forked a more recent hidapi](https://github.com/cboulay/hidapi/commits/38681c734965636c9fd710dc3898d90aaf923bb9) and made some changes to it such that it now works in multiple Windows versions and OS X.

I also made a change to the number of bytes hidapi returns in Windows to reverse a problematic commit made about 8 months after the hidapi commit pointed to currently by thp/psmoveapi/external/hidapi. Obviously this commit was made for a reason, but it causes the number of bytes returned to be different in Windows and OSX. I don't know if this is specific to Win10 only.

thp, 2016-12-01:
@cboulay Did you try to submit your changes to https://github.com/signal11/hidapi via pull request?

cboulay, 2016-12-01:
No. The most important change was done by someone else, committed on [March 3](https://github.com/signal11/hidapi/commit/b5b2e1779b6cd2edda3066bbbf0921a2d6b1c3c0).

The problem with hid_get_feature_report returning different numbers of bytes in Mac and Windows still exists in their repo. According to their [API docs](https://github.com/signal11/hidapi/blob/master/hidapi/hidapi.h#L312-L315), they want it to behave the Windows way where res = number of bytes in the feature report + 1. This is not what happens in Mac. I suppose I should make an issue about this.

Until this is fixed, I think the best course of action would be to have the submodule point to the signal11/hidapi repo, but then change all the `res == X` checks to `res == X || res == X+1`. I've even seen some hidapi using code that checks if `res >= X`. I don't know if this has the potential to be harmful, probably not.

cboulay, 2016-12-01:
There's already a pull request for it. https://github.com/signal11/hidapi/pull/219

rafaellago, 2016-12-26:
Will this issue be adressed in the 4.0.0??

thp, 2016-12-26:
It's probably not yet fixed in 4.0.0, but hopefully a point release after that (now that we have proper CI and build/release infrastructure in place, it should be easy to do more regular releases, since the cost of doing a release is basically just doing a Git tag).

@cboulay If I understand you correctly, if we replace the result checks with something like this, it should work, right? (the +1 only happens on Windows?)

```
#if defined(_WIN32)
#define PSMOVE_CHECK_HIDAPI_RESULT(value, expected) ((value) == ((expected) + 1))
#else
#define PSMOVE_CHECK_HIDAPI_RESULT(value, expected) ((value == (expected))
#endif
```

cboulay, 2016-12-26:
I know hidapi in MacOS returns one less byte than hidapi in Windows. According to the hidapi API docs, returning (report-size + 1) bytes is expected, so it almost makes more sense to have `((expected)-1)` for Mac, but that's just semantics.

This is a problem that should be addressed in hidapi itself, and I hope it will be soon, but your solution above should work until then.

I haven't tested in Linux.

rafaellago, 2017-07-13:
I recompiled the project using @cboulay hidapi, it seemed to work fine, at least now the binaries recognize the controller. But I'm having a real hard time to pair it to bluetooth. Don't actually know if it is another problem regarding W10.

commit 13b5e6a0 (2017-09-10): Merge pull request #319 from thp/win10-fixes-cboulay

hidapi: Use Windows 10 fixes by cboulay (Fixes #214)
closed 2017-09-10 by thp