thp/psmoveapi #222: Can't get windows 10 x64 to work.

greasysock, 2016-03-15:
I can't seem to get the ps camera working on windows 10.  I've installed the libusb drivers for the camera.

This is what I see when I initially load an example:

[image]

This is what I see when I run ps3eye_sdl:
[image]

So I know that it's working in a way. 

I can sort of make it work by first running tuio_server and then starting some tests, but it still dosn't work right

[image]

Any suggestions?

cboulay, 2016-03-16:
It's probably a bandwidth issue. ps3eye_sdl uses a low resolution so it doesn't tax the USB channel very much. Try unplugging other USB devices that are plugged in next to the ps eye.

Are you using Zadig to install the libusb drivers? Which version did you install? libusbK, win32, or WinUSB? While they all worked for me on Windows 8 and below, only one worked on Windows 10. I'm away from my Win10 machine right now and I can't remember which. It doesn't take long to try them all. Before you switch to a different driver, be sure to use control panel to uninstall the old one and also delete the driver file when you uninstall it (it's a checkbox option when you uninstall).

greasysock, 2016-03-18:
I tried every driver in Zadig. Pulled out everything but my keyboard. Discovered that the camera only works right when i have it in a usb 3.1 port. 

Then I fired up a lubuntu VM in virtualbox and passed through the camera and controller, was able to see the camera and run the examples fine. In Guvcview, I can only open up a 50fps stream at 640x480.

rovarma, 2016-03-21:
I got a new Win10 machine yesterday, so I tried this out. The camera still works fine for me (using libusb-win32), so I'm inclined to agree with @cboulay that this is a hardware issue.

thp, 2016-12-28:
Closing as hardware issue as per the latest two comments.

closed 2016-12-28 by thp