thp/psmoveapi #223: CMake won't work on Raspi

hdsjulian, 2016-03-23:
Using a raspberry pi 3 here with a freshly installed raspbian. Unfortunately I get the following error

`pi@raspberrypi:~/psmoveapi/build $ JAVA_HOME=/usr/lib/jvm/java-7-openjdk-armhf/
 cmake ..
CMake Error at src/CMakeLists.txt:182 (target_link_libraries):
  Cannot specify link libraries for target "psmoveapi_static" which is not
  built by this project.
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

-- Configuring incomplete, errors occurred!
See also "/home/pi/psmoveapi/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/psmoveapi/build/CMakeFiles/CMakeError.log".`

Any Ideas what to do?

thp, 2016-03-26:
Can you attach the whole logfile? (CMakeOutput.log/CMakeError.log)

hdsjulian, 2016-03-28:
pi@raspberrypi:~/psmoveapi/build $ cat /home/pi/psmoveapi/build/CMakeFiles/CMak
eError.log
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec3790104560/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3790104560.dir/build.make CMakeFiles/cmTryCompileExec3790104560.dir/build
make[1]: Entering directory '/home/pi/psmoveapi/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3790104560.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3790104560.dir/CheckSymbolExists.c.o   -c /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec3790104560
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3790104560.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec3790104560.dir/CheckSymbolExists.c.o  -o cmTryCompileExec3790104560 -rdynamic 
CMakeFiles/cmTryCompileExec3790104560.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x38): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec3790104560.dir/build.make:88: recipe for target 'cmTryCompileExec3790104560' failed
make[1]: Leaving directory '/home/pi/psmoveapi/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec3790104560] Error 1
Makefile:118: recipe for target 'cmTryCompileExec3790104560/fast' failed
make: *** [cmTryCompileExec3790104560/fast] Error 2

File /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec1982088716/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1982088716.dir/build.make CMakeFiles/cmTryCompileExec1982088716.dir/build
make[1]: Entering directory '/home/pi/psmoveapi/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/psmoveapi/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1982088716.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec1982088716.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec1982088716
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1982088716.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec1982088716.dir/CheckFunctionExists.c.o  -o cmTryCompileExec1982088716 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec1982088716.dir/build.make:88: recipe for target 'cmTryCompileExec1982088716' failed
make[1]: Leaving directory '/home/pi/psmoveapi/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec1982088716] Error 1
Makefile:118: recipe for target 'cmTryCompileExec1982088716/fast' failed
make: *** [cmTryCompileExec1982088716/fast] Error 2

thp, 2016-03-30:
`/usr/bin/ld: cannot find -lpthreads` -- do you have the pthreads development libraries installed? `libc6-dev` or something. You are probably missing other build-dependencies as well, please install them before.

thp, 2016-05-17:
I guess you got that working in the meantime or is it still an issue?

closed 2016-12-01 by thp