thp/psmoveapi #224: Raspi: Psmovepair won't work with several controllers?

hdsjulian, 2016-04-05:
Well basically this happens when I try and pair. I have a raspberry pi 3. 

Connecting the first controller: 
pi@raspberrypi:~/src/psmoveapi/build $ sudo ./psmovepair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in e0:ae:5e:57:70:aa not yet calibrated.
PSMove #1 connected via USB.
Pairing of #1 succeeded!
Controller address: e0:ae:5e:57:70:aa
Calibration data available and saved.

second one:
pi@raspberrypi:~/src/psmoveapi/build $ sudo ./psmovepair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 00:06:f7:15:22:71 not yet calibrated.
PSMove #1 connected via USB.
[PAIRING LINUX] Trying to stop bluetoothd...
[PAIRING LINUX] Using sysvinit...
[....] Stopping bluetooth (via systemctl): bluetooth.service.
[PAIRING LINUX] Succeeded
[PAIRING LINUX] Trying to start bluetoothd...
[PAIRING LINUX] Using sysvinit...
[....] Starting bluetooth (via systemctl): bluetooth.service.
[PAIRING LINUX] Succeeded
Pairing of #1 succeeded!
Controller address: 00:06:f7:15:22:71

third one:
pi@raspberrypi:~/src/psmoveapi/build $ sudo ./psmovepair
Connected controllers: 0

retry:
pi@raspberrypi:~/src/psmoveapi/build $ sudo ./psmovepair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 00:07:04:9b:e7:60 not yet calibrated.
PSMove #1 connected via USB.
WARNING: Can't determine Bluetooth address.
Make sure Bluetooth is turned on.
Pairing of #1 failed.
Calibration data available and saved

what am I doing wrong?

thp, 2016-05-17:
Have you checked if you have Bluetooth turned on?

thp, 2016-12-27:
This should be fixed with: https://github.com/thp/psmoveapi/pull/294

closed 2016-12-27 by thp