thp/psmoveapi #226: Simple Python problems

hdsjulian, 2016-04-12:
Hi there, 

so i'm getting the simplest of python problems and wondering why. 

calling pair.py from the examples hits: 

pi@raspberrypi:~/src/psmoveapi/examples/python $ python pair.py
Traceback (most recent call last):
  File "pair.py", line 39, in <module>
    if move.connection_type == psmove.Conn_Bluetooth:
TypeError: in method 'PSMove_connection_type_get', argument 1 of type 'PSMove *'

rest of my problems seems based on this kind of behaviour. 

>>> moves = [psmove.PSMove(x) for x in range(psmove.count_connected())]
>>> moves
[<psmove.PSMove; proxy of None >, <psmove.PSMove; proxy of None >]

hdsjulian, 2016-04-13:
OK, short update: all api calls result in TypeErrors. I have no Idea why.

rafaellago, 2016-04-13:
Have you tried importing the so file directly through ctypes? On windows I had for more success through this way. I'm at work now, so I can't confirm that this is truly possible on the RPI. But if I find some time tonight, I'll run some tests.

hdsjulian, 2016-04-13:
Thanks, please do and as this goes beyond my python capabilities please provide a line of code or 2 :)

rafaellago, 2016-04-14:
Ok, it took me a while because I had to build on my Pi (and I suck at building native code). Here is some commands I threw at the python console. One thing I noticed, it was that I had to run as sudo to get my controller recognized, and haven't paired on bluetooth because I'm on a USB shortage. 

Hope this helps. 

```
pi@raspberrypi:~/psmove/psmoveapi-master/build $ sudo python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from ctypes import *
>>> moveapi = cdll.LoadLibrary("/home/pi/psmove/psmoveapi-master/build/libpsmoveapi.so.3")
>>> moveapi
<CDLL '/home/pi/psmove/psmoveapi-master/build/libpsmoveapi.so.3', handle 14061f0 at b6a1fbb0>
>>> moveapi.psmove_init(0x030901)
1
>>> controller = moveapi.psmove_connect()
[PSMOVE WARNING] Magnetometer in 00:06:f7:ac:3f:f5 not yet calibrated.
>>> moveapi.psmove_set_leds(controller, 255, 0, 0)
21124240
>>> moveapi.psmove_update_leds(controller)
1
```

Using ctypes, you have to call the C methods just like they are defined on psmove.h file. Its quite simple. 

Here is a python tutorial on how it works.
https://docs.python.org/3/library/ctypes.html

Hope this helps you.

hdsjulian, 2016-04-14:
WHAT. so actually i feel dumb now. Or just misinformed. The problem: running python as root even makes the python bindings work. So actually it all works now. thanks a lot.

rafaellago, 2016-04-15:
You are welcome. It took me while to find out that things had be run as root too. Don't worry.

hdsjulian, 2016-04-15:
Todo for psmoveapi: write some proper doc. Will do at some point.

thp, 2016-05-17:
Feel free to contribute the docs at some point. Closing for now, as the issue itself has been resolved, and just needs to be properly documented.

closed 2016-05-17 by thp