thp/psmoveapi #227: Kernel Panic

hdsjulian, 2016-04-15:
I am currently trying to debug a game called oust. It is available on github here 
https://github.com/Jonty/Oust
the main script is a python file: https://github.com/Jonty/Oust/blob/master/oust.py

My problem is that on Raspberry PI 3 I get a massive amount of kernel panics whilst trying to pair controllers. In all I'm trying to pair several. 
No Idea how exactly. I think I can reproduce it by pulling out the cable after more than 5 seconds but it happens in any kind of stage. Sometimes with Controller 1, sometimes with another. 
Unfortunately Kernel Panics aren't easy to capture so here's a screenshot. I'm grateful for any kind of hint. 

[image]

thp, 2016-05-17:
Have you tried with a newer kernel version or different Bluetooth dongle (are you using a BT dongle or is Bluetooth built-in there)?

adangert, 2017-09-02:
Looks like this is still an issue for pairing controllers, using a bt dongle.

adangert, 2017-09-14:
looks like your issue might be here:
https://github.com/Jonty/Oust/blob/master/oust.py#L75

This was just solved in Joustmania: https://github.com/adangert/JoustMania/blob/master/piparty.py#L420

if move.get_serial() is called on a move that was just pulled out then the kernal will panic.

thp, 2017-09-15:
@adangert Note that with the new API, there's now a way to get notified (on Linux and macOS, Windows support is not implemented yet for this feature) of controller connects and disconnects at runtime, using the new API (https://github.com/thp/psmoveapi/blob/master/include/psmoveapi.h and https://github.com/thp/psmoveapi/blob/master/bindings/python/psmoveapi.py).

closed 2020-01-04 by nitsch