thp/psmoveapi #228: psmovepair.exe Failed to find a Bluetooth radio

Cman1468, 2016-05-02:
I cant get the controllers to connect with my PC. When I run psmovepair.exe the window just stays open for an instant, then closes. When I run the exe in cmd, then this is what I get:

 E:\VR\SteamVR\OSVR_PSMove\PSMove>psmovepair.exe
 Connected controllers: 1
 [PSMOVE WARNING] Magnetometer in 00:06:f7:1c:09:d1 not yet calibrated.
 PSMove #1 connected via USB.
 [PSMOVE WARNING] Failed to find a Bluetooth radioPairing of #1 failed.
 Calibration data available and saved.

It looks like it can't find my bluetooth radio, but mines plugged in. Why can't it find it? All of the drivers are installed, and it connects to my phone just fine. This is the one I have: http://www.amazon.com/TRENDnet-Bluetooth-Distance-Compatible-Classic/dp/B003KU4DOG?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00
I also tried the bluetooth radio from this set: http://www.amazon.com/Panda-Wireless-Combo-Wifi-Bluetooth/dp/B00C5PJHNY/ref=sr_1_6?s=pc&ie=UTF8&qid=1461977705&sr=1-6&keywords=raspberry+pi+bluetooth Also didn't work. I'm also running Windows 10, if that makes a difference

thp, 2016-05-02:
Do you use the normal Windows 10 Bluetooth stack or a third party software solution? The pairing code only supports the Windows 10 stack, but you can specify the adapter's address as command-line argument to psmovepair.

Cman1468, 2016-05-02:
OMG that did it!! Thanks man!

closed 2016-05-02 by thp