thp/psmoveapi #231: Cross-compiling missing libusb-1.0

zelmon64, 2016-05-10:
Hi,

I tried to cross-compile for Windows 10 on Ubuntu 14.04.4 LTS and found that I had to download the binaries for libusb-1.0 and place the MinGW versions in the /external/libusb-1.0 folder. Could this be implemented automatically in the /contrib/build_scripts/debian/cross-compile-mingw64 file or documented somewhere please?

Cheers

rafaellago, 2016-12-21:
Have you tried again with a more recent build of the psmove api? I built it yesterday without trouble. 

And are you trying to run the script from the root dir of the project and as root? I had a lot of trouble before finding out that I had to sudo the build script. After that, compiled ok

zelmon64, 2016-12-21:
@rafaellago Thanks for the suggestions. I can't remember how I did it but I no longer need to tinker with it since the release of PSMoveService (I was using it in a FreePIE plugin to play with SteamVR). Possibly I just downloaded the zip file rather than doing a recursive clone like you say. I'll close this issue then.

closed 2016-12-21 by zelmon64