thp/psmoveapi #232: Controller and PS eye don't work together if Controller is connected only via bluetooth

DoctorZurk, 2016-05-16:
Hi, first of all, I want to thanks all the contributors of this great library.

I compiled successfully latest git branch with the Mac(10.11.4) helped by the info found here:
[https://docs.google.com/document/d/16oB5jpbCpeRWb0o7w0b9bFbT3vEzWGqEPESWZUSCPuU/edit]

The MotionController is pairing and working great, upon issuing: ./example.
Light on, vibration on, everything is fine.
The PS eye camera was a little trickier, but works, I can see the 640x480 at 60fps.

I could even play together great with the Opengl2 or the motion track examples, but only if the controller was plugged in the USB port.

As soon as the controller is connected via bluetooth, the tracker examples (any executable that uses the camera also) hang (don't crash) until I disable the bluetooth. When I disable bluetooth, the openGL windows are starting again rendering. If I enable bluetooth again while going, the windows freeze (spinning color circle) until I disable bluetooth again.

With controller connected via USB was working all a charm.

What the issue can be ???

thp, 2016-05-16:
Hm, are you sure it's not the other way round (working with Bluetooth-only, hanging with USB)? We can't yet read the input reports via USB, so if trying to read sensor values from USB, it hangs, but reading input values from Bluetooth works. However, setting the LEDs should work in both USB and Bluetooth modes.

DoctorZurk, 2016-05-16:
Yes I confirm that talking only to the controller works both with USB and bluetooth only.
But when the motion tracker comes in play, the examples hang until I disable bluetooth.
I have just make a test now, with the test_tracker example, was all ok with usb plugged controller, as soon as I switched on Bluetooth (device not yet connected) the GL window hanged, and when I disabled again the bluetooth, restarted rendering.
So yes, I'm sure, motion tracker works only with controller connected via usb at the moment.
The device is paired, if I click on the bluetooth button of the controller, immediately connects, and the leds and rumble examples all work.

DoctorZurk, 2016-05-17:
Found out what the problem is.
The problem is indeed with the PS3Eye and bluetooth controller.
PS3Eye works only with a controller connected over usb.
I tried disable PS3Eye in the configuration ccmake for using the built-in Facetime HD Camera,
now both controllers work over bluetooth and tracking works (not so good than on PS3Eye with cable though).
What can I do to use the PS3Eye instead ? Compile it 32bit only specifying OSX_Architecture: x86 ?

DoctorZurk, 2016-05-17:
With the integrated camera of the iMac, was all going great tracking 2 bluetooth paired PS move but, when I put the controllers on the table, I got this:

```
2016-05-17 15:17:01.717 test_tracker[539:41518] -[CaptureDelegate doFireTimer:]: unrecognized selector sent to instance 0x7fb55a404430
2016-05-17 15:17:01.725 test_tracker[539:41518] An uncaught exception was raised
2016-05-17 15:17:01.725 test_tracker[539:41518] -[CaptureDelegate doFireTimer:]: unrecognized selector sent to instance 0x7fb55a404430
2016-05-17 15:17:01.726 test_tracker[539:41518] (
	0   CoreFoundation                      0x00007fff8aa554f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff8b16773c objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff8aabf1ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
	3   CoreFoundation                      0x00007fff8a9c5571 ___forwarding___ + 1009
	4   CoreFoundation                      0x00007fff8a9c50f8 _CF_forwarding_prep_0 + 120
	5   Foundation                          0x00007fff8f92cdec __NSFireTimer + 95
	6   CoreFoundation                      0x00007fff8a9d2b94 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
	7   CoreFoundation                      0x00007fff8a9d2823 __CFRunLoopDoTimer + 1075
	8   CoreFoundation                      0x00007fff8a9d237a __CFRunLoopDoTimers + 298
	9   CoreFoundation                      0x00007fff8a9c9871 __CFRunLoopRun + 1841
	10  CoreFoundation                      0x00007fff8a9c8ed8 CFRunLoopRunSpecific + 296
	11  Foundation                          0x00007fff8f8a7ed9 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 270
	12  Foundation                          0x00007fff8f93bfc3 -[NSRunLoop(NSRunLoop) runUntilDate:] + 108
	13  libpsmoveapi_tracker.3.dylib        0x0000000103b65a31 _ZN12CvCaptureCAM9grabFrameEd + 241
	14  libpsmoveapi_tracker.3.dylib        0x0000000103b58cc1 cvGrabFrame + 17
	15  libpsmoveapi_tracker.3.dylib        0x0000000103b4f5e1 camera_control_query_frame + 33
	16  libpsmoveapi_tracker.3.dylib        0x0000000103b50639 psmove_tracker_update_image + 57
	17  test_tracker                        0x0000000103b36248 main + 504
	18  libdyld.dylib                       0x00007fff977645ad start + 1
	19  ???                                 0x0000000000000001 0x0 + 1
)
2016-05-17 15:17:01.726 test_tracker[539:41518] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CaptureDelegate doFireTimer:]: unrecognized selector sent to instance 0x7fb55a404430'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8aa554f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff8b16773c objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff8aabf1ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
	3   CoreFoundation                      0x00007fff8a9c5571 ___forwarding___ + 1009
	4   CoreFoundation                      0x00007fff8a9c50f8 _CF_forwarding_prep_0 + 120
	5   Foundation                          0x00007fff8f92cdec __NSFireTimer + 95
	6   CoreFoundation                      0x00007fff8a9d2b94 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
	7   CoreFoundation                      0x00007fff8a9d2823 __CFRunLoopDoTimer + 1075
	8   CoreFoundation                      0x00007fff8a9d237a __CFRunLoopDoTimers + 298
	9   CoreFoundation                      0x00007fff8a9c9871 __CFRunLoopRun + 1841
	10  CoreFoundation                      0x00007fff8a9c8ed8 CFRunLoopRunSpecific + 296
	11  Foundation                          0x00007fff8f8a7ed9 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 270
	12  Foundation                          0x00007fff8f93bfc3 -[NSRunLoop(NSRunLoop) runUntilDate:] + 108
	13  libpsmoveapi_tracker.3.dylib        0x0000000103b65a31 _ZN12CvCaptureCAM9grabFrameEd + 241
	14  libpsmoveapi_tracker.3.dylib        0x0000000103b58cc1 cvGrabFrame + 17
	15  libpsmoveapi_tracker.3.dylib        0x0000000103b4f5e1 camera_control_query_frame + 33
	16  libpsmoveapi_tracker.3.dylib        0x0000000103b50639 psmove_tracker_update_image + 57
	17  test_tracker                        0x0000000103b36248 main + 504
	18  libdyld.dylib                       0x00007fff977645ad start + 1
	19  ???                                 0x0000000000000001 0x0 + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

thp, 2016-05-17:
Which OpenCV version did you use to build PS Move API? Have you tried PSEYEDriver for interfacing with the PSEye on OS X (note that your backtrace says "cvGrabFrame", which suggests OpenCV)?

cboulay, 2016-05-17:
@DoctorZurk I'm guessing your PS Eye problem has to do with USB bandwidth issues. The PS3EYEDriver version of PS Eye takes up a lot of bandwidth and it's probably blocking out your bluetooth dongle, assuming you're using a dongle. There are many reports of similar problems in the past. Try moving your bluetooth dongle to another USB port on a different controller.

DoctorZurk, 2016-05-17:
No the bluetooth is built-in the iMac I'm using.
Ok it has only USB 2.0 (iMac 2011) but the controllers are now both working via bluetooth if I compile for the built-in camera (Facetime HD Camera).
Is really weird.

@thp I installed openCV with brew install brew/science/opencv command linking with ready binary (if I remember well) so I guess the latest (of the v2.x branch ?). 
Of course I tried the PS3EyeDriver and disabled the Code Laboratory driver (which is only for windows). 

I was guessing that I should compile opencv for x86 architecture to have it working ?

Stil is very weird, that, with the controller plugged to usb, the PS3Eye camera works and the Tracker works !

DoctorZurk, 2016-05-18:
I could try to compile psmoveapi against the 32bit version of OpenCV and see how it goes.
Can anybody point me in the right direction to do so ?

DoctorZurk, 2016-05-19:
I have simply put i386 in the OSX_ARCHITECTURE in the ccmake config, but this doesn't seem to get a OpenCV 32bit version. Infact the PS3Eye camera, works great, but just alone (or with controller on usb).

thp, 2023-01-03:
Is this still an issue?

thp, 2023-01-03:
Closing for now, please reopen if it's still an issue.

closed 2023-01-03 by thp