thp/psmoveapi #247: Latest build automation fails on Windows 10 Visual Studio 2013 [LINKER ERROR] [SNPRINTF]

DoctorZurk, 2016-05-25:
I finally paired the controllers and have the PS3 eye working on my win10 old box.

I tried to build on Windows 10 with Visual Studio 2013.
Followed the latest info on building, launched finally from the cloned repo directory the 2013 batch.
All went fine (config and build of the externals) but when compiling the library it failed with linker error, all about the snprintf missing reference externals.

What should I do to fix this ?

DoctorZurk, 2016-05-26:
I fixed it by subs snprintf calls to _snprintf.

thp, 2016-05-26:
@rovarma Is there an easy way to have AppVeyor build both VS2015 (as it does right now) and VS2013 as separate builds with separate results, so that we can spot VS2013 build errors more easily?

DoctorZurk, 2016-05-26:
I could finally run the openGL tests and everything.

One problem I noticed is with USB bandwidth issues. 
Maybe this can help others.
USB has just one bus on most of computers, the more USB devices you plug, the less bandwith you will get. I could finally have the PS3eye working when almost all devices were disconnected from the computer (only  mouse, keyboard, bluetooth dongle, camera).

I'm wondering now how to use MOVED to send out OSC (over udp) data from the controllers, the help file is very short, I'm a missing something in that ?
Are there existing tuio implementations to send out controllers data ?

mirh, 2016-07-10:
@thp Yes there is. 
I know pcsx2 does this for example.

thp, 2016-12-28:
Accepting patches for either making AppVeyor build the VS2013 solution as well, or for fixing the code so that VS2013 builds it. For now, closing this as WONTFIX (use VS2015, which we test regularly and provide builds for).

closed 2016-12-28 by thp