thp/psmoveapi #248: Random disconnects

spoonymushroom, 2016-06-10:
I'm experiencing random disconnections when the Move controller is connected via Bluetooth on my Macbook running OS X 10.11.5. After it disconnects, I can't reconnect the controller by pressing the PS button. I can get it working again if I connect the controller via USB, and the LED starts flashing, then pressing the PS button. I'm having this issue with multiple controllers.

Also, I've noticed that the LED doesn't turn solid when the controller is paired. It keeps flashing.

Any ideas how I can get to the bottom of this?

nitsch, 2016-06-13:
I am not a Mac user, so I cannot help you with debugging this. But I
will say this:

> Also, I've noticed that the LED doesn't turn solid when the 
> controller is paired. It keeps flashing.

This means that there is no working Bluetooth connection. Which probably
means that you are no dealing with random disconnects but rather with
problems to connect at all.

spoonymushroom, 2016-06-14:
Even when the light's blinking, the Move appears in my Bluetooth devices and my apps using the API and the sample apps all work until the random disconnection happens. 

@nitsch So if I'm having "problems to connect", as you say, what would you recommend as a way to debug this?

nitsch, 2016-06-14:
Like I said, I am no Mac user. Maybe @cboulay or @thp can help you with
that.

nitsch, 2016-06-14:
Although ... If you are building the library from source, you could
build with -DPSMOVE_USE_DEBUG=1 to enable additional debug output. Maybe
that gives you a clue.

thp, 2016-06-14:
If the connection works and the light keeps blinking, this might indicate that it needs to be charged (i.e. the battery gets empty). I think it does start to blink when the battery is < 20% full.

cboulay, 2016-06-14:
I've never had either of my controllers randomly disconnect from my mac unless they run out of battery. I guess that it's either a problem with your OS (faulty bluetooth stack, power saving?) or it's a hardware problem.

BTW, there are multiple flashing modes. Slow flashing is charging. Faster flashing is "I'm trying to pair". I can't remember if it flashes very fast when it's about to die (like the DS3 does). Slow flashing when you're plugged in is totally normal. If you leave it plugged in overnight and it's still flashing, then your batteries aren't holding a charge. If it disconnects whenever you remove the USB cable then the batteries aren't holding a charge.

This type of bad-battery behaviour has happened to many others, especially with older/used PSMove controllers. One person told me they hacked their USB internals of the controller to directly apply a 5V charge to the internal battery, then after a full charge like that they were able to use it normally. To me it seems like these controllers are cheap enough that it would be easier to buy a new one.

spoonymushroom, 2016-06-14:
You may be right that it's a battery issue. Although the API reports a battery level of 2 / 5 (which I thought has okay), the light was blinking quickly. Another problem was that the controllers don't seem to charge from a USB power socket (no slow blinking when plugged in).

Thanks for all your help!

closed 2016-06-14 by spoonymushroom