thp/psmoveapi #251: undefined reference to `cvRound' in examples & tests with -DPSMOVE_USE_DEBUG=1

ChristophHaag, 2016-08-15:
```
[ 83%] Linking C executable test_tracker
CMakeFiles/test_record_video.dir/examples/c/test_record_video.c.o: In function `cvPointFrom32f':
/usr/include/opencv2/core/types_c.h:929: undefined reference to `cvRound'
/usr/include/opencv2/core/types_c.h:930: undefined reference to `cvRound'
CMakeFiles/test_record_video.dir/examples/c/test_record_video.c.o: In function `cvReadInt':
/usr/include/opencv2/core/core_c.h:2318: undefined reference to `cvRound'
CMakeFiles/test_record_video.dir/examples/c/test_record_video.c.o: In function `cvEllipseBox':
/usr/include/opencv2/imgproc/imgproc_c.h:1041: undefined reference to `cvRound'
/usr/include/opencv2/imgproc/imgproc_c.h:1042: undefined reference to `cvRound'
```
Apparently it has something to do with opencv 2 vs opencv 3. I have opencv 3.1.0 here on Archlinux.

It still builds with
```
cmake -DPSMOVE_USE_DEBUG=1 -DPSMOVE_BUILD_EXAMPLES=0 -D PSMOVE_BUILD_OPENGL_EXAMPLES=0 -DPSMOVE_BUILD_TESTS=0
```
so it's not critical for debugging, but still annoying

thp, 2016-08-31:
Has OpenCV 3 removed those functions? Or put into another header?

nitsch, 2016-12-27:
See https://github.com/opencv/opencv/issues/6076, https://github.com/opencv/opencv/issues/6585.

suryaprakaz, 2017-01-27:
Is the only way to get around this issue either using OpenCV2 with C or use OpenCV3 with C++?

nitsch, 2017-02-13:
> Is the only way to get around this issue either using OpenCV2 with C
> or use OpenCV3 with C++?

Well, it looks as if that is the case. To be honest, I do not even know
why the update to OpenCV 3 has been made for PS Move API. Should have
sticked to 2.x ...

thp, 2022-09-17:
I just tried building in ubuntu 22.04 using:

```sh
scripts/build_package.sh
```

This builds OpenCV 3.4 from source, and PS Move API successfully links against it.

The affected test application has been removed in 2016 already: https://github.com/thp/psmoveapi/pull/277

closed 2022-09-17 by thp