thp/psmoveapi #252: Doesn't actually pair on Archlinux

ChristophHaag, 2016-08-15:
bluez 5.41

```
sudo ./psmovepair
Connected controllers: 1
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
[PSMOVE WARNING] Magnetometer in 00:06:f5:20:07:97 not yet calibrated.
PSMove #1 connected via USB.
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
[PSMOVE DEBUG] Already paired.
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
[PAIRING LINUX] Trying to stop bluetoothd...
[PAIRING LINUX] Using systemd...
[PAIRING LINUX] Succeeded
[PAIRING LINUX] Trying to start bluetoothd...
[PAIRING LINUX] Using systemd...
[PAIRING LINUX] Succeeded
Pairing of #1 succeeded!
[PSMOVE DEBUG] Current host: 00:30:91:40:08:5d
Controller address: 00:06:f5:20:07:97
Calibration data available and saved.
```
Creates those files

/var/lib/bluetooth/00:30:91:40:08:5D/00:06:F5:20:07:97/info
```
[General]
Name=Motion Controller
Class=0x002508
SupportedTechnologies=BR/EDR;
Trusted=true
Blocked=false
Services=00001124-0000-1000-8000-00805f9b34fb;

[DeviceID]
Source=1
Vendor=1356
Product=981
Version=1
```

and /var/lib/bluetooth/00:30:91:40:08:5D/cache/00:06:F5:20:07:97
```
[General]
Name=Motion Controller

[ServiceRecords]
0x00010000=3601920900000A000100000900013503191124090004350D35061901000900113503190011090006350909656E09006A0901000900093508350619112409010009000D350F350D350619010009001335031900110901002513576972656C65737320436F6E74726F6C6C65720901012513576972656C65737320436F6E74726F6C6C6572090102251B536F6E7920436F6D707574657220456E7465727461696E6D656E740902000901000902010901000902020800090203082109020428010902052801090206359A35980822259405010904A101A102850175089501150026FF00810375019513150025013500450105091901291381027501950D0600FF8103150026FF0005010901A10075089504350046FF0009300931093209358102C0050175089527090181027508953009019102750895300901B102C0A1028502750895300901B102C0A10285EE750895300901B102C0A10285EF750895300901B102C0C0090207350835060904090901000902082800090209280109020A280109020B09010009020C093E8009020D280009020E2800
```

but bluetoothctl says
```
[NEW] Controller 00:30:91:40:08:5D c-l [default]
[NEW] Device 00:06:F5:20:07:97 Motion Controller
[bluetooth]# info 00:06:F5:20:07:97
Device 00:06:F5:20:07:97
        Name: Motion Controller
        Alias: Motion Controller
        Class: 0x002508
        Icon: input-gaming
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v054Cp03D5d0001
```

I did get it paired and working with a different bluetooth adapter some time ago by just trying a couple of times but right now it doesn't get paired at all. Maybe the config file syntax changed a bit again?

nitsch, 2016-08-15:
> Maybe the config file syntax changed a bit again?

Yes, maybe. Pairing is very hackish for Linux and thus likely to break
whenever BlueZ changes anything (as it happened with BlueZ 5). Maybe
there are better, more high-level ways for doing this now ...

Does the OS report anything when you press the PS button on the
controller after successfully running psmovepair?

ChristophHaag, 2016-08-15:
Since it's not really paired, no, it doesn't report anything.

ChristophHaag, 2016-08-15:
Wait. I just found my bluetooth applet saying the controller is connected, but not paired. So I disconnected it and clicked on connect again - and it connected and works fine. Meanwhile bluez says this:
```
Motion Controller]# info 00:06:F5:20:07:97
Device 00:06:F5:20:07:97
        Name: Motion Controller
        Alias: Motion Controller
        Class: 0x002508
        Icon: input-gaming
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v054Cp03D5d0001
```
It's connected, but not paired? What the heck?
I think it's buggy behavior of bluez.

remi6397, 2017-02-14:
Happens to me too, on Arch Linux (bluez 5.43); yesterday on the same build it was working good.

thp, 2017-02-24:
Did you check if your hidraw device permissions are correct? Your user must have read/write permissions on `/dev/hidraw*` (at least the nodes that represent the PS Move controller).

remi6397, 2017-02-24:
Thank you for response, but I'm a bit confused... My Move refuses to pair and I cannot connect it to PC. 
This is my generic Bluetooth controller:
```
ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```
This is pairing log:
```
# ./psmove pair
Connected controllers: 1
PSMove #1 connected via USB.
[PAIRING LINUX] Trying to stop bluetoothd...
[PAIRING LINUX] Using systemd...
[PAIRING LINUX] Succeeded
[PAIRING LINUX] Trying to start bluetoothd...
[PAIRING LINUX] Using systemd...
[PAIRING LINUX] Succeeded
Pairing of #1 succeeded!
Controller address: 00:07:04:29:c7:ef
Calibration data available and saved.
```
This is output from bluetoothctl:
```
--- cut ---
[NEW] Device 00:07:04:29:C7:EF Motion Controller
[bluetooth]# info 00:07:04:29:C7:EF
Device 00:07:04:29:C7:EF
	Name: Motion Controller
	Alias: Motion Controller
	Class: 0x002508
	Icon: input-gaming
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v054Cp03D5d0001
```
Controller works properly when connected to USB.

ChristophHaag, 2017-04-09:
Well, it just randomly worked for me. Now I can't make it work. So let's reopen this.

closed 2020-10-12 by nitsch