thp/psmoveapi #253: OSX Instructions Not Quite Right

francoislaberge, 2016-09-06:
**OS X El Capitan (10.11.5)**

I just got my playstation move gear and am trying out this project. I'm running into some discrepancies in the [OSX Google Doc Instructions](https://docs.google.com/document/d/16oB5jpbCpeRWb0o7w0b9bFbT3vEzWGqEPESWZUSCPuU/edit) on my computer. Here are a few things that don't quite square up:
  1. OS X contribs are not where the docs insinuate.  This `bash -e -x contrib/build-osx-snapshot` should be `bash -e -x contrib/build_scripts/osx/build-osx-snapshot`
  2. After correcting for the above I have issues with cmake finding the SDL headers:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
SDL2_INCLUDE_DIR
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi
   used as include directory in directory /Users/francoislaberge/dev/psmoveapi

-- Configuring incomplete, errors occurred!
See also "/Users/francoislaberge/dev/psmoveapi/build/CMakeFiles/CMakeOutput.log".
```

I stopped at that as my hacking through it just led to further problems. Any help would be appreciated.

francoislaberge, 2016-09-06:
Note sure if this matters either, but the example used `/Developer/Library/` as the example installation location, but that folder doesn't exist naturally on at least my version of OSX, I have xcode installed already if that matters.

rovarma, 2016-09-14:
Sorry for the late reply. I'm not sure where you got the Google Docs link from, but it is quite outdated. The correct instructions for installing/building on all platforms is here: https://github.com/thp/psmoveapi/blob/master/docs/build.rst

From your output, I would guess that SDL2 is not installed on your system. Please try following the updated instructions and see if that helps.

francoislaberge, 2016-09-14:
Wow, now I'm curious where I got it too. Could have sworn it was by reading the top level docs. If I figure out my confusion point I'll submit a suggestion. In the meantime, thanks!

cboulay, 2016-09-15:
I just tried a fresh clone/build following those instructions and came upon the following error:

```
 53%] Built target psmoveapi_tracker
[ 54%] Linking CXX executable test_opengl2
Undefined symbols for architecture x86_64:
  "_SDL_ClearError", referenced from:
      checkSDLError(int) in test_opengl2.cpp.o
      Renderer::Renderer(Tracker&) in test_opengl2.cpp.o
  "_SDL_CreateWindow", referenced from:
      Renderer::Renderer(Tracker&) in test_opengl2.cpp.o
  "_SDL_GL_CreateContext", referenced from:
      Renderer::Renderer(Tracker&) in test_opengl2.cpp.o
  "_SDL_GL_SwapWindow", referenced from:
      Renderer::render() in test_opengl2.cpp.o
      Main::exec() in test_opengl2.cpp.o
  "_SDL_GetError", referenced from:
      sdlDie(char const*) in test_opengl2.cpp.o
      checkSDLError(int) in test_opengl2.cpp.o
      Renderer::Renderer(Tracker&) in test_opengl2.cpp.o
  "_SDL_Init", referenced from:
      Renderer::Renderer(Tracker&) in test_opengl2.cpp.o
  "_SDL_PollEvent", referenced from:
      Main::exec() in test_opengl2.cpp.o
  "_SDL_Quit", referenced from:
      sdlDie(char const*) in test_opengl2.cpp.o
      Renderer::~Renderer() in test_opengl2.cpp.o
      Renderer::~Renderer() in test_opengl2.cpp.o
      _main in test_opengl2.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test_opengl2] Error 1
make[1]: *** [CMakeFiles/test_opengl2.dir/all] Error 2
make: *** [all] Error 2
```

 I added `-DPSMOVE_BUILD_OPENGL_EXAMPLES=OFF` to the flags in build-osx-snapshot then the build was able to complete, but obviously this is not the desired solution.

TimArt, 2016-11-23:
I am also on OS X El Capitan and got past this error by installing SDL2 on my system.
If you've got homebrew:
`brew install sdl2`

Only problem now for me is that I'm getting errors in some of the source code in src/tracker.

I believe this is a problem because I replaced the opencv external library with OpenCV 3 so I think some of the older code is depreciated :(.
I was forced to use OpenCV 3 because of an earlier build error where OpenCV 2 required Apple's  QTKit library which has been depreciated in El Capitan & Xcode 8 for Apple's AVFoundation library.

So now, I think I'm either going to need to rewrite some of the source to work with OpenCV 3 or go back to OpenCV 2 and make it work with AVFoundation (but the second option sounds way harder I think).

nitsch, 2016-11-23:
I just wanted to suggest checking the setup for automatic OSX builds on [Travis](https://travis-ci.org/thp/psmoveapi/builds) since they are also on El Capitan, but these builds seem to have also been broken some time ago (see #261). Maybe you can check the build/error logs on Travis and also compare the changes that broke the build to see if that has anything to do with your problem. And, of course, finally submitting fixes would be great!

TimArt, 2016-11-24:
I just made a pull request that now has the API working for me on El Capitan. The current version of opencv is the main problem I was running into. It includes Apple's depreciated QTKit library. The replacement library Apple has now fully adopted is AVFoundation. Whenever I was building, it would error out on opencv for this reason. An unmerged pull request has been created for opencv that includes this AVFoundation functionality. I patched this into the mac osx bash buildfile: build-osx-snapshot.
Still seems to crash on TravisCI, not really sure why. I'll try to investigate that further.

thp, 2016-11-26:
Closing this in favor of #263 -- let's continue the discussion there.

closed 2016-11-26 by thp