thp/psmoveapi #254: Cant compile examples using gcc in archlinux

feztrath, 2016-09-22:
Hello I would like to inform a problem i had using the library i'm using archlinux kernel 4.4.21-1-lts gcc version 6.2.1 20160830

i tried compiling the library it did with a lot of warnings on implicit declarations and pthreads (not pthread) but it worked, also compiled the examples in the examples and the C ones worked i tried running the python ones and also worked, then i tried creating a 'C' script to compile it and try it out so i included "psmove.h" didnt work, then "psmoveapi/psmove.h" however i didn't included functions from the library yet because i was testing the includes and compiled successfully a hello world with the "psmoveapi/psmove.h" include in it however i wanted to test the library so i grabed some of  the code on one of the examples and tried to compile it, didn't work now here i thinking that i may have messed up my code i decided to compile the example.c  using gcc however it gave me the error "#include "psmove.h" no such file or directory" then i renamed it to "#include "psmoveapi/psmove.h" then said 

"../../src/psmove_port.h:32:20: error: psmove.h: No such file or directory
 #include "psmove.h"

then i renamed the "psmove.h" in the "psmove_port.h" to be "psmoveapi/psmove.h" and it gave me 

"reference to (insert name of function here) not defined" for each of the functions 
the file has my first few tries where typing 

"gcc example.c" 

after searching a bit 

"gcc -lpsmoveapi example.c"

neither worked, even so the c examples compiled with make worked as runing the python examples as root btw i also tried sudo with the 'C' examples 

if possible i will like to know what to do in this situation, also how can i use the C# samples i dont know whom i should try to compile, and if i can call them from the shared library using either language

Im kinda new in the Software Development, Git and Linux world so help will be appreciated, also if i have typo, gramatic or sintactic errors it because i talk spanish natively, thanks to those who read and all for your help :)

nitsch, 2016-09-30:
So what exactly is your problem now? I find it hard to follow.

Are you having problems compiling your own program using the psmoveapi library?

feztrath, 2016-09-30:
yes whenever i try to compile a program in C using the psmoveapi library gives me errors

a tried yesterday to compile a program using it, however i can compile if i just 
include the library "psmoveapi/psmove.h" but  i dont use any function, yesterday also let me declare a variable of type "PSMove" (PSMove *move) and compiled.

Even thought i can include the library and define a variable using the "psmoveapi/psmove.h" library whenever i call a function like "psmove_connect();" for example and compile, it gives me   the "Undefined reference to 'psmove_connect' " error

here the code (i used the example.c file as reference, and finally erased almost everything for testing purposes) : 

#include <stdio.h>
#include <stdlib.h>
#include <psmoveapi/psmove.h>
#include <time.h>
#include "GLFW/glfw3.h"

int main (int argc, char* argv[])
{
	PSMove *move;
	enum PSMove_Connection_Type ctype;

	int i;
	move = psmove_connect();
	printf("Hello World");
	return 0;
}

the Error (its in spanish since i use a spanish version of archlinux) :

gcc trial.c

/tmp/ccvzfFON.o: En la función `main':
trial.c:(.text+0x15): referencia a `psmove_connect' sin definir
collect2: error: ld devolvió el estado de salida 1

in eglish will be something like:

/tmp/ccvzfFON.o: in function `main':
trial.c:(.text+0x15): undefined reference to `psmove_connect'
collect2: error: ld returned exit status 1

it seems its trying to find the function in the current trial.c script but i dont know why wont recognize only functions from the psmove library

i checked the libraries and the so but all looks in place they are in the /usr/local folder in the "include" and also in the "lib64" even the "bin" with the "psmovepair, magnetometer_calibration and moved" binaries

PD: also tried to compile various examples from the Examples/C folder but it gives me the same "undefined function" errors

cboulay, 2016-09-30:
You seem to be including the "psmove/psmove.h" header correctly, otherwise you'd get an error sooner.
I think you are forgetting to **link** the psmove .so. Try [this answer](http://stackoverflow.com/a/726390/1256069) on stackoverflow.

feztrath, 2016-09-30:
Thanks, i tried it but no luck here what i tried and what it gave me

[image]

not sure if i did it right, or if i left something out

nitsch, 2016-09-30:
You simply need to pass -lpsmoveapi, i.e. *without* the prefix "lib".
This will search for the library in standard system directories and any
directories that you optionally provided via the -L option.

  gcc trial.c -lpsmoveapi

feztrath, 2016-09-30:
thanks @nitsch  but didnt work also i think i place this in the first message anyway heres my output

this happened:

First:
[image]

Second:
[image]

Third:
[image]

nitsch, 2016-09-30:
a.out is the default name if you do not provide one via the -o option.

The error you get when running the executable says that the shared
library was not found at runtime. To fix this, try one of the solutions
suggested here:

https://www.gnu.org/software/gsl/manual/html_node/Shared-Libraries.html

feztrath, 2016-09-30:
Thanks @nitsch  I Finally got the solution it ran, if i have any other troubles i may ask again.

I'll close it now 

Thank you very much @nitsch  and @cboulay . :smile:

closed 2016-09-30 by feztrath