thp/psmoveapi #256: Static lib not installed

nitsch, 2016-10-03:
The static lib is built but not part of the install targets.

thp, 2016-10-15:
That's correct, and I guess is the intended way so that by default, distribution packages link against the shared library and not accidentally link against the static library. If somebody wants to statically link against PS Move API, they probably build it with a specific configuration and link it in their build system and not from the system installation directory.

If you feel that it's useful to also install the static library, I'm happy to accept a pull request :)

thp, 2016-12-22:
Closing for now, only the shared library should be installed, the static library should only be used in custom builds, not installed system-wide.

closed 2016-12-22 by thp