thp/psmoveapi #260: No bluetooth adapter detected!

chrisbgn, 2016-11-23:
Hello guys, i have a problem with the bluetooth adaptar that i recently purchased and it is the following: When i plug in my TRENDnet TBW-106UB bluetooth adapter and i turn it on from sys tray task bar it seems to work pretty well but when i start psmoveservice.exe then psmoveconfigtool.exe the program cant find any bluetooth adapter, here's the log:

[2016-11-22 22:39:57.776]: main - Starting PSMoveService
[2016-11-22 22:39:57.785]: bluetooth_get_host_address - Failed to find a bluetooth radio
[2016-11-22 22:39:57.797]: PSMoveController::open - Opening PSMoveController(\\?\hid#vid_054c&pid_03d5&col01#7&3080afd0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030})
[2016-11-22 22:39:57.800]: PSMoveController::open -   with EMPTY serial_number
[2016-11-22 22:39:57.809]: DeviceTypeManager::update_connected_devices - Device device_id 0 (PSMove) opened
[2016-11-22 22:39:57.853]: PS3EyeTracker::open - Opening PS3EyeTracker(USB\VID_1415&PID_2000\6\0, camera_index=0)
[2016-11-22 22:39:58.60]: SharedMemory::initialize() - Allocating shared memory: tracker_view_0
[2016-11-22 22:39:58.370]: DeviceTypeManager::update_connected_devices - Device device_id 0 (PSEYE) opened
[2016-11-22 22:39:58.372]: PS3EyeTracker::open - Opening PS3EyeTracker(USB\VID_1415&PID_2000\7\1, camera_index=1)
[2016-11-22 22:39:58.557]: SharedMemory::initialize() - Allocating shared memory: tracker_view_1
[2016-11-22 22:39:58.622]: DeviceTypeManager::update_connected_devices - Device device_id 1 (PSEYE) opened

I have win 7 64-bit, and drivers installed. Any help would be thankful!

cboulay, 2016-11-23:
You should probably close this issue and instead open one in [psmoveservice](https://github.com/cboulay/PSMoveService/issues).

chrisbgn, 2016-11-23:
ok, thank you

closed 2016-11-23 by chrisbgn