thp/psmoveapi #262: Do not merge broken builds

nitsch, 2016-11-23:
Having the automated builds on [Travis](https://travis-ci.org/thp/psmoveapi) and [AppVeyor](https://ci.appveyor.com/project/thp/psmoveapi) does not help much if errors in these builds are ignored and problematic pull requests are merged anyway.

Merging into master should ideally be automated (with the built-in requirement to only consider branches that build on all platforms) or at least be preceded by "manual" checks of results on Travis and AppVeyor.

thp, 2016-11-26:
Yes, that should be the case. It seems like the failing builds were due to some weirdness in Travis' Mac OS X environment, and I was just too impatient to fix the (seemingly environment-related) bug. Especially since the builds that failed in Travis worked great locally for me. I tried to fix it some time ago ("travis" branch), but it didn't help. I will give it another look.

thp, 2016-11-26:
Oh, seems like the build error in this case is not related to the OS X failures. My fault! :)

thp, 2016-11-26:
The builds are fixed again... for now.

closed 2016-11-26 by thp