thp/psmoveapi #264: Library Linking Issues

TimArt, 2016-11-25:
## Library Linking Error
I've had a similar library linking error occur multiple times in different contexts.
It's usually one of the two errors:
`ld: library not found for -libpsmoveapi`
or
`ld: library not found for -lpsmoveapi`

One problem is that there might be a naming issue. It seems the most recent build names the library files that are build in /build as "libpsmoveapi . . ." but in older makefiles is is sometimes referred to as "lpsmoveapi . . . ". 

## Triggers of the Error
These kinds of errors are triggered for me in two situations:

1. When I try to write a basic C example that uses the library, Xcode gives me the error:
`ld: library not found for -lpsmoveapi_tracker.3.1.0`
It seems that Xcode is looking for the library titled: "lpsmoveapi_tracker.3.1.0" as opposed to the library titled: "libpsmoveapi_tracker.3.1.0" which I have included as a library in my Xcode project.
Is there some kind of build file I can change that is telling XCode to use this older name?

2. When I try to build the "labs" in /examples/labs. It seems that some of these examples use QT. I get the same error as above:
`ld: library not found for -libpsmoveapi`
For the paint lab I was able to modify the paint.pro QT project file to get it to sucessfully bulild, but I am not very familiar with QT or syntax to know how I should be properly linking the library.

If anybody can give me a bit more information about linking libraries and using build files, I'd like to try to fix this.
Thanks!

thp, 2016-11-26:
The library file is named `libpsmoveapi.3.1.0.dylib` (with symlinks to it called `libpsmoveapi.3.dylib` and `libpsmoveapi.dylib`). The linker flag used to link against it is `-lpsmoveapi` (it's normal that the `-l` flag doesn't include the "lib" prefix). The compiler flag `-libpsmoveapi` is _wrong_, it should be `-lpsmoveapi`. If you get an error `ld: library not found for -lpsmoveapi` you have to fix your library search path (e.g. `-L/path/to/where/the/dynlib/file/is/`, note the uppercase "L" for search path).

TimArt, 2016-11-27:
Thanks, that was my fault, I'm learning this for the first time and forgot to type in the library search paths to the build settings. Thank you!

thp, 2016-11-27:
Glad that you fixed it :) Closing.

closed 2016-11-27 by thp