thp/psmoveapi #283: Error on step 4

joeb456, 2016-12-22:
This keeps happening.

bash -e -x contrib/build-osx-snapshot
bash: contrib/build-osx-snapshot: No such file or directory

Should I make the folders contrib and build-osx-snapshot, or what?

_im writing this issue here because the mailing list seems impossible to join_

nitsch, 2016-12-22:
> bash -e -x contrib/build-osx-snapshot
> bash: contrib/build-osx-snapshot: No such file or directory
> 
> Should I make the folders contrib and build-osx-snapshot, or what?

No, you should use the correct path to the file like this:

  bash -e -x contrib/build_scripts/osx/build-osx-snapshot

This is what the build instructions in docs/build.rst state.

Where did you get that path you were using?

> _im writing this issue here because the mailing list seems impossible
> to join_

Please give some details. The list, as it is currently configured,
should be free to join for everyone.

joeb456, 2016-12-23:
1: It came from the readme.osx

 file that came with the api

> 4. Build all
> 	$ bash -e -x contrib/build-osx-snapshot

2: I found a link that said it was from one of the documentations I found (I now found the correct link).

Thanks for responding! (I'll look for the link that said it was the mailing list, and add a comment or something)

nitsch, 2016-12-23:
> 1: It came from the readme.osx
> 
>  file that came with the api
> 
>> 4. Build all
>> 	$ bash -e -x contrib/build-osx-snapshot

Are you referring to a release that you downloaded from GitHub? Version
3.9.1 has file readme.osx. But it instructs users to clone the latest
sources from GitHub and then refers to a file that has been moved after
the release. Hm, I think that needs to be fixed.

nitsch, 2016-12-23:
Yes, please close this issue if the suggested solution works for you.

joeb456, 2016-12-23:
Thanks for the help!

joeb456, 2016-12-23:
Ouch. I tried your code this morning and it didn't work either.
I'm going to try do this myself this time.
(https://github.com/thp/psmoveapi/blob/master/contrib/build_scripts/osx/build-osx-snapshot)

nitsch, 2016-12-23:
> Ouch. I tried your code this morning and it didn't work either.
> I'm going to try do this myself this time.
> (https://github.com/thp/psmoveapi/blob/master/contrib/build_scripts/osx/build-osx-snapshot)

You reopened the ticket, so I assume you want some help. Please give
some information on what exactly is not working, which version you are
trying to build (latest state of the GitHub repository or some official
source release) etc.

joeb456, 2016-12-23:
I'm getting the same error again, and I got psmoveapi from https://thp.io/2010/psmove/ .

> bash -e -x contrib/build_scripts/osx/build-osx-snapshot
> bash: contrib/build_scripts/osx/build-osx-snapshot: No such file or directory

nitsch, 2016-12-23:
> I'm getting the same error again, and I got psmoveapi from https://thp.io/2010/psmove/ .
> 
>> bash -e -x contrib/build_scripts/osx/build-osx-snapshot
>> bash: contrib/build_scripts/osx/build-osx-snapshot: No such file or directory

You must either clone from GitHub and use the path above ... or you
unpack the 3.9.1 release and use the path

  contrib/build-osx-snapshot

that is mentioned in the release's build instructions. Mixing both
things won't work.

joeb456, 2016-12-23:
Thanks! I'll test it now.

thp, 2016-12-28:
The docs are now up to date, and there's new build scripts. I've also removed the outdated reference to the Mac OS X docs.

closed 2016-12-28 by thp