thp/psmoveapi #288: Controller connects and immediatley disconnects

lmore377, 2016-12-23:
I have no idea what's going on, but if I run "./psmove pair" my controller pairs successfully, but when i turn it on it connects but immediately disconnects, but if I pair it manually with a usb bluetooth dongle, it works.

nitsch, 2016-12-25:
First of all: What platform are you on? Which version of the library are you using?

> but when i turn it on it connects but immediately disconnects

What exactly are you doing? Do you mean pressing the PS button on the controller to establish the Bluetooth connection?

> but if I pair it manually with a usb bluetooth dongle, it works.

What exactly are you doing? Do you mean calling `psmove pair` with a bluetooth address?

lmore377, 2016-12-25:
First: I'm on Ubuntu with the latest build, 2nd: I do mean that, and third:
I did do it with a Mac address

nitsch, 2016-12-26:
You can run CMake with an additional -DPSMOVE_USE_DEBUG=1 to enable
building a version with debug output. Recompile and run `psmove pair`
again. That should give you some information on the Bluetooth address
the controller is currently paired to. See if you can spot differences
between your two ways of doing the pairing.

lmore377, 2016-12-26:
Without mac address [Pairing to internal bluetooth card] (psmove pair)
`Connected controllers: 1
devs_sorted[0]: (handle=0x561dbf4027c0, serial=, path=/dev/hidraw0)
[PSMOVE DEBUG] Current host: 00:15:##:##:##:##
[PSMOVE DEBUG] Current host: 00:15:##:##:##:##
PSMove #1 connected via USB.
[PSMOVE DEBUG] Current host: 00:15:##:##:##:##
[PSMOVE DEBUG] Current host: c4:d9:##:##:##:##
[PAIRING LINUX] Trying to stop bluetoothd...
[PAIRING LINUX] Using upstart...
[PAIRING LINUX] Succeeded
[PAIRING LINUX] Trying to start bluetoothd...
[PAIRING LINUX] Using upstart...
[PAIRING LINUX] Succeeded
Pairing of #1 succeeded!
[PSMOVE DEBUG] Current host: c4:d9:##:##:##:##
Controller address: e0:ae:##:##:##:##
Calibration data available and saved.`

With mac address [External bluetooth dongle] (psmove pair 00:15:##:##:##:##)
`Using user-supplied host address: 00:15:##:##:##:##
Connected controllers: 1
devs_sorted[0]: (handle=0x558d64b19510, serial=, path=/dev/hidraw0)
[PSMOVE DEBUG] Current host: c4:d9:##:##:##:##
[PSMOVE DEBUG] Current host: c4:d9:##:##:##:##
PSMove #1 connected via USB.
[PSMOVE DEBUG] Current host: c4:d9:##:##:##:##
[PSMOVE DEBUG] Current host: 00:15:##:##:##:##
Pairing of #1 succeeded!
[PSMOVE DEBUG] Current host: 00:15:##:##:##:##
Controller address: e0:ae:##:##:##:##
Calibration data available and saved.`

If you haven't noticed, I put the #'s

nitsch, 2016-12-26:
You apparently have (at least) two different Bluetooth transceivers in
your computer. The one that `psmove pair` finds first and uses is not
the one you manually specify.

If there are problems establishing a Bluetooth connection to one but not
to the other, my guess would be that they are configured differently
(visibility, allowed connection modes etc.).

If you do not want to investigate this further, you could simply turn
off the one Bluetooth device that `psmove pair` finds, which should then
make it use the one you previously specified manually.

nitsch, 2016-12-26:
Apparently, `psmove pair` finds the internal transceiver first and uses that one. If there are problems establishing a Bluetooth connection to the internal one but not to the external one, my guess would be that they are configured differently (visibility, allowed connection modes etc.).

If you do not want to investigate this further, you could simply turn off the internal Bluetooth device, which should then make `psmove pair` use the external one whose address you previously provided manually.