thp/psmoveapi #301: Third PS Move never connects on Macs

underyx, 2017-01-23:
A friend and I've tried to connect three PS Move controllers to our MacBooks running Sierra. Regardless of which two we've started with, the third one we tried to connect never did. Pairing and registering seemed to be successful, but after disconnecting USB, we weren't able to connect it, while the first two controllers worked perfectly.

What info could I provide to help debug this?

thp, 2017-01-28:
I had the same problem after updating to macOS Sierra. Sometimes it's possible to connect more, but there's definitely something fishy going on, on the same hardware before Sierra it worked great with up to 7-9 controllers being connectable via Bluetooth.

fonograph, 2017-08-21:
It's working consistently fine for me with 6 controllers (don't have more to test) on 10.12.6. Maybe it's fixed in a recent update, or it might be the fact that I'm using an external Bluetooth dongle.

thp, 2017-08-26:
@fonograph it's still not working for me on 10.12.6. Which external Bluetooth dongle (brand + model name) are you using? I'd like to try it myself, that'd be an okay-ish workaround on macOS.

fonograph, 2017-08-28:
This is the one: https://www.startech.com/ca/Networking-IO/Bluetooth-Telecom/USB-Bluetooth-4-Dongle~USBBT1EDR4 <https://www.startech.com/ca/Networking-IO/Bluetooth-Telecom/USB-Bluetooth-4-Dongle~USBBT1EDR4>

thp, 2017-08-30:
I've ordered a Bluetooth USB dongle and connected it to my Mac today (running 10.12.6) and can verify that more than 3 controllers can be indeed connected (note that you need to pair while the dongle is connected, the internal Bluetooth will be disabled while the dongle is connected).

thp, 2017-09-10:
So I guess the "fix" here is that we document this issue and also how to work around it (by getting an external Bluetooth 4.0/4.1 dongle).

nitsch, 2020-01-05:
Documented in 69db49f.

closed 2020-01-05 by nitsch