thp/psmoveapi #302: Do I need bluetooth to use the psmove?

LegendL3n, 2017-02-13:
I have 2 PS Move controllers but neither of them work, I connect them(one at a time) to a 3.0 USB port and when I [run the test_tracker.exe](http://postimg.org/image/bjsbysi8j/)
When I run the "psmove pair" command I get [this](http://postimg.org/image/vgy97r13n/)

LegendL3n, 2017-02-13:
I also have [ScpToolkit ](https://github.com/nefarius/ScpToolkit) installed don't know if it interferes...

cboulay, 2017-02-13:
The PS Move doesn't actually communicate very much information over USB.
USB is only useful for pairing and a few other things. It needs bluetooth.

I don't know about ScpToolkit but if it takes control of the PSMove's
bluetooth interface then yes it will interfere. Though I've never tried
personally, HipsterSloth uses ScpToolkit for the PS Navi support so the
software doesn't have to interfere. But he did go out of his way to say you
need to unplug/disable the bluetooth used for the PSMove's before
installing ScpToolkit. See instructions here:
https://github.com/cboulay/PSMoveService/wiki/PSNavi-Setup

LegendL3n, 2017-02-13:
I have a ps move not navi but ok I prob need a bluetooth dongle right?

cboulay, 2017-02-13:
Sorry I didn't look carefully at the source of the notification and I thought this was for a different library. My answer remains the same, but I should be less authoritarian and "I'm not going to support this" when I'm outside my own repo.

I suppose there are some things you could do with USB-only. You can set the bulb LEDs via USB-only so if all you want to do is track a sphere then you don't need bluetooth. Controller orientation data and button functionality definitely require bluetooth.

Not all Bluetooth dongles work. I and many others have had pretty good success with the ASUS BT-400.

LegendL3n, 2017-02-13:
Though all I want to hook is the led to show light

rafaellago, 2017-02-13:
For the leds, and vibration you're good with cable. 

Never tested much, and other people more involved with the project could tell you better. But I think the ps move have all the functionality through USB, the drawback is you have limited movement due to the cable being attached.

LegendL3n, 2017-02-13:
But why the error then?

rafaellago, 2017-02-13:
PSMove pair is to pair the controller via Bluetooth, if you wanna fiddle with leds, you don't have to run this command. Try testing with `psmove test-led-pwm-frequency`. It should make your ps move light up momentarily

cboulay, 2017-02-13:
@rafaellago , you do not have all the functionality through USB. Orientation and button presses do not work through USB.

@LegendL3n , It's normal to get one or two libusb errors when it's trying to connect to the camera. How many webcams do you have plugged in?

LegendL3n, 2017-02-13:
@cboulay I have a PSEye but it isn't connected

@rafaellago it says I need the controller connected via Bluetooth so I'll have to buy a dongle

But what's strange is that when I run "psmove battery" works and the controller's LED(or light ball) lights up purple so I suppose I can hook it without a dongle if I code myself something right?

cboulay, 2017-02-13:
@LegendL3n 

test_tracker will fail if you don't have a camera connected and installed properly. The "tracker" concept means to track the bulb position using a camera. Installing a PSEye in Windows can be a little tricky. See instructions [here](https://github.com/cboulay/PSMoveService/wiki/PSEye-Software-Setup-(Windows)). 

Also, "it says I need the controller connected via Bluetooth..." **What** says that? If you're referring to the `psmove pair` error then yeah you're going to get an error when you're trying to pair -- which is a bluetooth specific concept -- with no bluetooth. That doesn't mean you need a dongle for everything, but you definitely need one to perform bluetooth pairing.

LegendL3n, 2017-02-13:
@cboulay "psmove led-pwm-frequency"

LegendL3n, 2017-02-13:
@cboulay and I have my pseye working with cl eye

cboulay, 2017-02-13:
Maybe the led-pwm-frequency setting requires bluetooth. I can't remember. But you can write a program that uses psmoveapi to set the LED colour without bluetooth.

I don't know if @thp already dropped support for CL Eye but he said he was thinking about it. If that happens then you'll have to use ps3eyedriver.

LegendL3n, 2017-02-13:
@cboulay I don't really want to deal with the camera, just the psmove's LED
However I don't know how to include the API in my Visual Studio project...

cboulay, 2017-02-13:
[Here](https://psmoveapi.readthedocs.io/en/latest/) for instructions on how to build psmoveapi.
Then it's included and linked just like any other C API. I can't really go beyond that because there are many many other docs/tutorials on the internet on how to link a 3rd party API in your application.

LegendL3n, 2017-02-13:
@cboulay true but that way of building gives me errors

cboulay, 2017-02-13:
And what are the errors? Please be very detailed.

LegendL3n, 2017-02-13:
http://postimg.org/image/l2lzlp1eb/

cboulay, 2017-02-13:
At the bottom, click on the Output tab (instead of Error List tab). Highlight everything. Copy+Paste here.
Surround the pasted text with triple-backtick (```) to make it easier to read.

nitsch, 2017-02-13:
> Though all I want to hook is the led to show light 

You should rather try "example" (or "example_new_api" if you want C++)
then. All the other test applications cater very specific purposes. None
of which you are really interested in.

"example" also works with USB. It lights up the LEDs. So you can test
that everything is working and then take that code and work from there.

thp, 2017-02-24:
Looks like this question has been answered? You can use USB for pairing + LED (and rumble) setting, no sensor reading works over USB at the moment, we don't know how to make the controller report this data over USB, so you need Bluetooth for those features.

closed 2017-02-24 by thp