thp/psmoveapi #303: Please document how to build for windows

wirelessdreamer, 2017-02-16:
I'm used to doing dev under Linux, but need to build and modify this library under windows. when I try and do a cmake build in cygwin on windows it complains about a missing libudev library, which I know isn't used on windows. can you please add build instructions to the readme for windows. Thanks

wirelessdreamer, 2017-02-16:
I found build instructions here, https://github.com/thp/psmoveapi/blob/master/docs/build.rst
but the build process fails.

E:\src\psmoveapi>call scripts/visualc/build_msvc_2015.bat
Applying libusb dynamic CRT patch...

Building libusb
Microsoft (R) Build Engine version 14.0.24730.2
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: E:\src\psmoveapi\scripts\visualc\..\..\\external\libusb-1.0/msvc/libusb_static_2015.vcxproj
Failed to build libusb

There was an error running one of the build commands. Please see the output for more information.

The above error means nothing to me

nitsch, 2017-02-17:
The documentation states that Windows builds require Visual Studio. You
cannot just throw the build script at Cygwin and expect it to work.

Either use Visual Studio (2013 or 2015 Community editions) on Windows or
cross-compile for Windows on Linux, using MinGW.

BTW, a more readable representation of the docs can be found here:

  https://psmoveapi.readthedocs.io/en/latest/build.html

They are automatically generated from the .rst files in the docs/ directory.

thp, 2017-02-24:
It still works on CI, maybe have a look at how https://github.com/thp/psmoveapi/blob/master/.appveyor.yml does it?

wirelessdreamer, 2017-03-03:
I tried it again last night and it finished without issue. I'm not sure what I was doing wrong, thanks for the help.

closed 2017-03-03 by wirelessdreamer