thp/psmoveapi #307: Why Can't calibrate the magnetometer?

AvadaX, 2017-02-27:
I built PA MOVE API on Ubuntu 14.04,and it works well, except the magnetometer value is always 0

After built the api and paired, I tried to calibrate the magnetometer by: (in the build folder)
`sudo ./psmove calibrate`
and is shows
[image]
but no matter how I rotate the controller, the value is always 0, so I continued
[image]

After this, when I run any examples, it always show:
`[PSMOVE WARNING] Magnetometer in e0:ae:5e:e1:54:82 not yet calibrated.`
and the magnetometer value is 0.0, like this:
[image]

So, could someone help me with this ? How to make the magnetometer works and get the value?

cboulay, 2017-02-27:
Is this a new controller? Apparently the new PSMove controllers do not transmit magnetometer values, at least not the same as the old ones. I don't even know if they have a magnetometer. @HipsterSloth attempted but was unable to set the new version of the controller into the mode that allows firmware updates; he had hoped that an older firmware would enable magnetometer value transmission.

nitsch, 2017-02-27:
> Is this a new controller? Apparently the new PSMove controllers do not
> transmit magnetometer values, at least not the same as the old ones. I
> don't even know if they have a magnetometer. [...]

Interesting ... I dissected the firmware quite a bit (back then). I would
love to hear some details about problems with these new controllers.

AvadaX, 2017-02-28:
@cboulay 
yes, I am using the new controller (PS4 move controller). so it's the new controller' firmware problem,  OK, I will test old controller later. 

Thank you ~

closed 2017-02-28 by AvadaX

HipsterSloth, 2017-02-28:
@nitsch Sorry I meant to reply to this earlier today but I couldn't find the original thread that talked about this. Finally found it. This first came up with newer PSMoveControllers that came bundled with the PSVR (model: CECH-ZCM1U). 

The full thread can be found here:
https://groups.google.com/forum/#!searchin/psmoveservice/magnetometer|sort:relevance/psmoveservice/QDZLT0rWNyQ/An-6WZ5vBQAJ

TL;DR - It appears that the magnetometer values from byte 0x26-0x29 are zero out. Here is what a typical report should look like:
```
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

typical report:
01 00 00 00 06 00 00 7f 7f 7f 7f c0 05 2d 89 a8
7f 77 8d 1e 89 a0 7f 6f 8d f8 7f 02 80 fb 7f f8
7f 03 80 fb 7f 7d 10 9e fa 10 55 0a 00 00 00 00
00
```

And here is what we were getting from CECH-ZCM1U model controllers (I repro'd this issue with this model controller as well):

https://groups.google.com/d/msg/psmoveservice/QDZLT0rWNyQ/EtWnvIySBwAJ
```
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

at rest 1
01 00 00 00 04 00 00 7f-7f-7f-7f 46 03 70-7f 64-
90 89-7f 77-7f 5a-90 7d-7f f9-7f fc-7f 0e-80 02-
80 fd-7f 08-80 836  000 000  000 12 00 00 00 00
00

at rest 2
01 00 00 00 05 00 00 7f-7f-7f-7f 4a 03 73-7f 72-
90 77-7f 76-7f 68-90 7f-7f fc-7f fb-7f 04-80 00-
80 fa-7f 07-80 836  000 000  000 76 00 00 00 00
00

shaking about 1
01 00 00 00 00 00 00 7f-7f-7f-7f 73 03 45-77 d4-
86 d0-86 42-76 d5-85 01-86 e2-93 c9-7c 75-6a f9-
93 30-7c 3b-6a 845  000 000  000 84 00 00 00 00
00

shaking about 2
01 00 00 00 03 00 00 7f-7f-7f-7f 80 03 58-75 84-
82 15-80 5c-75 dd-84 04 7d 0b-92 2a-7a 46-68 74-
92 20-7a 00-68 845  000 000  000 b9 00 00 00 00
00

trigger, circle, select angle 1
01 01 20 108 a ff ff 7f-7f-7f-7f 38 03 d5-80 0d-
82 72-90 e2-80 00-82 78-90 03-80 de-7f 18-80 f9-
7f e4-7f 0c-80 836  000 000  000 3a 00 00 00 00
00

trigger, circle, select angle 2
01 01 20 108 3 ff ff 7f-7f-7f-7f aa 03 92-82 cb-
80 73-90 57-82 b2-80 6f-90 d0-7f 76-7f f4-7f e1-
7f 89-7f 0e-80 838  000 000  000 a9 00 00 00 00
00

green light on angle 1
01 00 00 000 3 00 00 7f-7f-7f-7f ce 03 d9-88 b6-
7f 8a-8d d1-88 af-7f 7c-8d 02-80 f7-7f 09-80 01-
80 f9-7f 11-80 848  000 000  000 4e 00 00 00 00
00

green light on angle 2
01 00 00 000 7 00 00 7f-7f-7f-7f 59 03 14-76 cf-
7f c3-8d 10-76 d1-7f b2-8d 04-80 f6-7f 0e-80 07-
80 fb-7f 08-80 848  000 000  000 32 00 00 00 00
00
```

I was hoping to try and make a utility to set the DFU mode over USB and then copy older firmware onto the newer controller, but it looks like the [feature report 0xf1](https://github.com/nitsch/moveonpc/wiki/Feature-report-0xF1) doesn't work when I try to send it using HidAPI. 

My failed attempt at setting the DFU mode was here:
https://github.com/cboulay/PSMoveService/commit/3cacd056cdea5a9a137e955a7d076f5d66634674#diff-5b702d89eda81e114954896af282f91fR920

nitsch, 2017-02-28:
Thanks. I will try to get my hands on one of these new controllers and
find some time to look into this. Chances are they rebuilt parts of it.
Anyone ever took one of the new controllers apart to have a first glance
(spot ICs etc.)?

nitsch, 2017-03-13:
> This first came up with newer PSMoveControllers that came bundled with
the PSVR (model: CECH-ZCM1U).

Isn't that just the old model? Searching for that model number, I find all
sorts of move controllers, dating back several years which is very likely
earlier than PSVR.

What model numbers do your old controllers have?