thp/psmoveapi #314: No module named '_psmove'?

adangert, 2017-08-27:
after building successfully on windows, running pair.py gives the following error:

```
C:\Users\Aaron\psmoveapi\examples\python>python pair.py
Traceback (most recent call last):
  File "pair.py", line 35, in <module>
    import psmove
  File "..\..\build\psmove.py", line 18, in <module>
    _psmove = swig_import_helper()
  File "..\..\build\psmove.py", line 17, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_psmove'
```

adangert, 2017-08-27:
output for the build is here: https://github.com/adangert/JoustMania/blob/Windows_support/build_output.txt

thp, 2017-08-28:
It looks like MSVC puts the files in different directories:
```
C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build\Release\_psmove.pyd
```

Try adding `C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build\Release` to your `%PYTHONPATH%` or copy `_psmove.pyd` in the same folder where `psmove.py` is.

adangert, 2017-08-28:
looks like that worked for getting past that error:
I am running windows 10, and it looked like psmove.exe pair was not pairing the psmove controllers correctly:
```C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build\Release>psmove.exe pair
Connected controllers: 1
Error connecting to PSMove #1
```

however https://github.com/cboulay/psmove-ue4/blob/master/Binaries/Win64/psmove-pair-win.exe
seemed to be able to connect my controller to the pc

still am unable to get the other programs to work correctly:
```
C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build\Release>psmove.exe list
Host address: 00:15:83:d1:1c:51
[PSMOVE CRITICAL] Assertion fail in psmove_get_serial: move != NULL
```
psmove.exe then crashes in the above example.

```
C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\examples\python>python pair.py
Traceback (most recent call last):
  File "pair.py", line 39, in <module>
    if move.connection_type == psmove.Conn_Bluetooth:
TypeError: in method 'PSMove_connection_type_get', argument 1 of type 'PSMove *'
```

nitsch, 2017-08-28:
With no Windows 10 machine currently available for testing I cannot try
this. But my guess would be this issues

  https://github.com/thp/psmoveapi/issues/214

which specifically affects Windows 10 and is probably still unresolved.

nitsch, 2017-09-19:
psmoveapi was recently updated, including the Windows 10 issue mentioned above. Have you tried version 4.0.4 to see if it solves your problem?

nitsch, 2017-10-07:
@adangert I just built release 4.0.4 and also the older 4.0.1 on Windows 10, tried pairing, BT connection, and example programs. I could not find any problems. Since your actual problem apparently has nothing to do with Python, I just skipped that for now.

I do, however, remember problems with BT disconnects that can lead to strange states on Windows. See this thread on the mailing list: [error after disconnect on windows](https://groups.google.com/d/msg/psmove/njN8hoIaWKM/26PJk7dP0EoJ)

So I tried that. After disconnecting a controller by pressing and holding the PS button for a while, Windows still listed this controller as connected BT device. Running `psmove list` and `example` produced exactly the same output you show in your previous post. Also the crash. So I am pretty confident that this is the actual problem.

The only thing you can do about it right now is to remove the controller from Windows's list of Bluetooth devices and cleanly reconnect via BT. Does that help?

adangert, 2017-10-09:
When I get a chance to try it again, I'll see if it works.

nitsch, 2017-12-02:
@adangert Any news on this?

thp, 2018-06-01:
Closing as fixed. Please re-open if you still experience this issue.

closed 2018-06-01 by thp