thp/psmoveapi #327: Document x64 build instructions - fatal error LNK1112: module machine type 'x64' conflicts with target machine type 'X86'

wirelessdreamer, 2017-09-13:
to reproduce:
using VS2017 community edition

After meeting all the dependencies and doing the cmake build open ALL_BUILD.vcxproj
leaving the default platform set to win32, right click the solution, click build, and a the build completes 8/8 successful tasks

right click the solution, go to properties, change the platform to x64. Then on the top menu click on build->configuration manager, and change the targets to x64 if not already set.
right click the solution and select clean solution. Then right click the solution and select build solution.
2 of 8 tasks finish successfully.

from the below build log it shows some of the project is still targeted at 32 bit. I'm not very familiar with this build environment, so I assume i'm missing changing a target setting from 32 to 64 bit. I testing some changes to the library, and need a 64 bit build of it to test.

Build Log: 

1>------ Build started: Project: ZERO_CHECK, Configuration: Debug x64 ------
1>Checking Build System
1>CMake does not need to re-run because D:/VR/psmoveapi/cmake/build/CMakeFiles/generate.stamp is up-to-date.
2>------ Build started: Project: psmoveapi, Configuration: Debug x64 ------
3>------ Build started: Project: psmoveapi_static, Configuration: Debug x64 ------
3>D:\Programs\Visual Studio\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(387,5): warning MSB8028: The intermediate directory (x64\Debug\) contains files shared from another project (psmoveapi.vcxproj).  This can lead to incorrect clean and rebuild behavior.
2>Building Custom Rule D:/VR/psmoveapi/CMakeLists.txt
3>Building Custom Rule D:/VR/psmoveapi/CMakeLists.txt
3>CMake does not need to re-run because D:/VR/psmoveapi/cmake/build/CMakeFiles/generate.stamp is up-to-date.
2>CMake does not need to re-run because D:/VR/psmoveapi/cmake/build/CMakeFiles/generate.stamp is up-to-date.
2>hid.c
3>hid.c
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(398): warning C4456: declaration of 'res' hides previous local declaration
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(272): note: see declaration of 'res'
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(398): warning C4456: declaration of 'res' hides previous local declaration
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(272): note: see declaration of 'res'
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(638): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(644): warning C4245: '=': conversion from 'int' to 'DWORD', signed/unsigned mismatch
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(655): warning C4245: '=': conversion from 'int' to 'DWORD', signed/unsigned mismatch
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(681): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(735): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(751): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(757): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(779): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(780): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(821): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(834): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(847): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
2>D:\VR\psmoveapi\external\hidapi\windows\hid.c(860): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
2>psmove.c
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(638): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(644): warning C4245: '=': conversion from 'int' to 'DWORD', signed/unsigned mismatch
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(655): warning C4245: '=': conversion from 'int' to 'DWORD', signed/unsigned mismatch
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(681): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(735): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(751): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(757): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(779): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(780): warning C4267: 'function': conversion from 'size_t' to 'DWORD', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(821): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(834): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(847): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
3>D:\VR\psmoveapi\external\hidapi\windows\hid.c(860): warning C4267: 'function': conversion from 'size_t' to 'ULONG', possible loss of data
3>psmove.c
3>D:\VR\psmoveapi\src\psmove.c(2183): warning C4244: 'return': conversion from 'uint64_t' to 'long', possible loss of data
3>D:\VR\psmoveapi\src\psmove.c(2220): warning C4013: 'mkdir' undefined; assuming extern returning int
2>D:\VR\psmoveapi\src\psmove.c(2183): warning C4244: 'return': conversion from 'uint64_t' to 'long', possible loss of data
2>D:\VR\psmoveapi\src\psmove.c(2220): warning C4013: 'mkdir' undefined; assuming extern returning int
2>psmove_calibration.c
3>psmove_calibration.c
3>moved_client.c
2>moved_client.c
3>D:\VR\psmoveapi\src\daemon\moved_client.c(111): warning C4996: 'inet_addr': Use inet_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
3>C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\um\winsock2.h(1830): note: see declaration of 'inet_addr'
3>D:\VR\psmoveapi\src\daemon\moved_client.c(116): warning C4996: 'gethostbyname': Use getaddrinfo() or GetAddrInfoW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
3>C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\um\winsock2.h(2218): note: see declaration of 'gethostbyname'
[... 47 lines cut ...]
3>psmove_alignment.cpp
3>psmove_quaternion.cpp
3>Generating Code...
3>x64\Debug\hid.obj : fatal error LNK1112: module machine type 'x64' conflicts with target machine type 'X86'
3>Done building project "psmoveapi_static.vcxproj" -- FAILED.
7>------ Build started: Project: psmove, Configuration: Debug x64 ------
7>D:\Programs\Visual Studio\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(387,5): warning MSB8028: The intermediate directory (x64\Debug\) contains files shared from another project (example.vcxproj, example_new_api.vcxproj, multiple.vcxproj, psmoveapi.vcxproj, psmoveapi_static.vcxproj).  This can lead to incorrect clean and rebuild behavior.
7>Building Custom Rule D:/VR/psmoveapi/CMakeLists.txt
7>CMake does not need to re-run because D:/VR/psmoveapi/cmake/build/CMakeFiles/generate.stamp is up-to-date.
7>psmovecli.cpp
7>d:\vr\psmoveapi\src\utils\moved.cpp(70): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
7>d:\vr\psmoveapi\src\utils\moved.cpp(188): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
7>d:\vr\psmoveapi\src\utils\moved.cpp(225): warning C4456: declaration of 'dev' hides previous local declaration
7>d:\vr\psmoveapi\src\utils\moved.cpp(207): note: see declaration of 'dev'
7>d:\vr\psmoveapi\src\utils\moved.cpp(302): warning C4458: declaration of 'output' hides class member
[... cut ...]
7>Done building project "psmove.vcxproj" -- FAILED.
8>------ Build started: Project: ALL_BUILD, Configuration: Debug x64 ------
8>Building Custom Rule D:/VR/psmoveapi/CMakeLists.txt
8>CMake does not need to re-run because D:/VR/psmoveapi/cmake/build/CMakeFiles/generate.stamp is up-to-date.
9>------ Skipped Build: Project: INSTALL, Configuration: Debug x64 ------
9>Project not selected to build for this solution configuration 
========== Build: 2 succeeded, 6 failed, 0 up-to-date, 1 skipped ==========

thp, 2017-09-13:
Did you try the `build_msvc.bat` script as outlined in http://psmoveapi.readthedocs.io/en/latest/build.html#building-on-windows-visual-studio? Building for 64-bit and 32-bit is supported, but right now, the easiest is to build it in different directories.

cboulay, 2017-09-13:
You'll probably want to use thp's script, but here's an explanation so you don't run into this problem in the future.

In general, any MSVC project you create via CMake will not support switching between 32-bit and 64-bit configurations because cmake does not tell the linker about both sets of dependent libraries, only the set of libraries that match the target at config time. By default, cmake configures 32-bit targets. If you want to configure a 64-bit target then you'll have to specify that via the [generator](https://cmake.org/cmake/help/latest/manual/cmake-generators.7.html#visual-studio-generators) and it will make an entirely separate set of project files from a 32-bit target.

wirelessdreamer, 2017-09-13:
I just tried build_msvc_2015.bat, but I get errors about build tools v140, even after installing the 2015 build tools package from microsoft.

I alsotried to create a 2017.bat with 2017 in the bat file in place of 2015, but it says only 2013 or 2015 are valid options.

thp, 2017-09-14:
If you see `build_msvc_2015.bat`, you have old code. The newest Git master branch has `build_msvc.bat` and supports VS2017. I've just recently tested this on Windows 10, you just need to install VS2017 Community (others probably work just as well) and make sure that you have "C++ Development" and "Windows 8.1 SDK" checked (even if you are on Windows 10), and then launch the `build_msvc.bat` with `2017` and the architecture you want (`x86` or `x64`) from the "Command prompt for Visual Studio 2017 Developer Tools" prompt (it's an item in the start menu after VS2017 is installed, the name might be slightly different).

thp, 2017-09-14:
..and once the build script is run once, you have `build-x86` and/or `build-x64` in your PS Move API checkout folder, and can open the project files there in Visual Studio.

nitsch, 2020-01-04:
No response from original poster. Assuming it is resolved.

closed 2020-01-04 by nitsch