thp/psmoveapi #328: Python & Raspberry Pi

Cynar, 2017-09-19:
I've been trying to get this working on the raspberry pi, in order to install 'oust' [(https://github.com/Jonty/Oust)](https://github.com/Jonty/Oust)

I've got it to compile and talk to the controllers, however, the game script fails with : 
```
Traceback (most recent call last):
  File "oust.py", line 4, in <module>
    import psmove
  File "/home/pi/psmoveapi/build/psmove.py", line 21, in <module>
    _psmove = swig_import_helper()
  File "/home/pi/psmoveapi/build/psmove.py", line 20, in swig_import_helper
    return importlib.import_module('_psmove')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_psmove)
```

The install mentions this issue: 
[https://github.com/thp/psmoveapi/pull/184](https://github.com/thp/psmoveapi/pull/184)

I couldn't use this though, since you seem to have modified the relevant code recently. Is there an obvious and easy fix I'm missing, or is it no longer as easy to patch?

Thanks

nitsch, 2017-09-19:
Which version of psmoveapi are you building? What does CMake output in its configure step?

I have no problems building and loading release 4.0.4 on Debian (currently version 9.1) which seems to be the base for Raspbian.

Cynar, 2017-09-19:
I'm building the latest version of psmoveapi, and using the raspberry pi zero w (armhf).

I've just reran the compile. Full log below. In doing so, I also realized I had to pull hidapi direct from their github as well as modify the JAVA_HOME line of the install instructions. When I compiled I got a load of warnings, but no errors and I was able to pair and calibrate a couple of the controllers.

Build and compile dump here: [https://pastebin.com/9wHB7bjC](https://pastebin.com/9wHB7bjC)

nitsch, 2017-09-20:
> In doing so, I also realized I had to pull hidapi direct from their github ...

No, you don't. Next time, simply clone with `--recursive` to initialize the submodules. Or call `git submodule init` and then `git submodule update` if you have already cloned (you might need to remove the hidapi clone you put there first).

> ... as well as modify the JAVA_HOME line of the install instructions.

Why? Since I assume you do not need Java bindings for oust, just build without it by adding `-DPSMOVE_BUILD_JAVA_BINDINGS=0` to the `cmake` call.

> Build and compile dump here: https://pastebin.com/9wHB7bjC

My guess right now would be that CMake finds another Python version than the one you are trying to run the result with. I can reproduce a similar error when compiling with Python 2.7 and running with Python 3.4.

To quickly verify that, replace the line
> set(INFO_BUILD_PYTHON_BINDINGS "Yes")

with

> set(INFO_BUILD_PYTHON_BINDINGS "Yes (version ${PYTHONLIBS_VERSION_STRING})")

in bindings/CMakeLists.txt and look for the version number in the output of CMake's configure step. What does it say?

nitsch, 2017-09-24:
@Cynar Any updates on this?

adangert, 2017-10-19:
we have the psmoveapi working on the raspberry pi, and also tested on the rpi-zero here:
https://github.com/adangert/JoustMania/blob/master/setup.sh

This might have been happening when running python2 so we switched to python3

thp, 2018-06-01:
Closing as fixed. Please reopen if you are still experiencing issues.

closed 2018-06-01 by thp