thp/psmoveapi #331: Build Issue on MacOS

araghuvanshi, 2017-11-16:
Tried building from the source code. Upon running bash -e -x scripts/macos/build-macos, I get scripts/macos/build-macos: line 33: ./autogen.sh: No such file or directory

thp, 2017-11-17:
Did you use `git clone --recursive` or `git submodule init && git submodule update`?

araghuvanshi, 2017-11-21:
Tried that and solved the lack of autogen.sh problem. However I have another issue:
[ 78%] Linking CXX shared library libpsmoveapi_tracker.dylib
ld: warning: text-based stub file /System/Library/Frameworks//QuartzCore.framework/QuartzCore.tbd and library file /System/Library/Frameworks//QuartzCore.framework/QuartzCore are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks//QuartzCore.framework/QuartzCore.tbd and library file /System/Library/Frameworks//QuartzCore.framework/QuartzCore are out of sync. Falling back to library file for linking.
ld: library not found for -lvtkRenderingOpenGL2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libpsmoveapi_tracker.dylib] Error 1
make[1]: *** [CMakeFiles/psmoveapi_tracker.dir/all] Error 2
make: *** [all] Error 2

I'm running 10.12.6 if that helps

closed 2017-11-27 by araghuvanshi